johang/sd-card-images #1: A13-Olinuxino-MICRO boot issue

ekawahyu, 2020-08-29:
I followed all instruction to make the SD card image and dd it into micro SD card, but here is what I got in continuous loop:
```U-Boot 2020.07 (Aug 27 2020 - 00:42:52 +0000) Allwinner Technology

CPU:   Allwinner A13 (SUN5I)
Model: Olimex A13-Olinuxino Micro
I2C:   ready
DRAM:  256 MiB

U-Boot SPL 2020.07 (Aug 27 2020 - 00:42:52 +0000)
DRAM: 256 MiB
CPU: 1008000000Hz, AXI/AHB/APB: 3/2/2
Trying to boot from MMC1
```

ekawahyu, 2020-08-31:
Closing this issue, since I am able to boot with a newly bought microSD card. There seems to be no issue when I use the same micro SD card for Raspberry Pi, but it just refuses to work with A13-Olinuxino-MICRO

closed 2020-08-31 by ekawahyu

johang, 2020-08-31:
Strange.

ekawahyu, 2020-09-01:
Yeah, very strange. I don't spend much time to investigate this. Anyway, @johang, this is a very cool project. I love this! I can't wait to test some of those images on different platform.

johang, 2020-09-02:
That's nice to hear. If the same image works with a different SD cards, I suspect it's a bug in u-boot somewhere.