johang/sd-card-images #2: Lichee Pi Zero not booting

ekawahyu, 2020-10-13:
@johang Recently I got Lichee Pi Zero, trying to boot it up, but no success. Any hint of what and where I need to look to solve this issue?
```
U-Boot SPL 2020.10 (Oct 12 2020 - 23:08:36 +0000)
DRAM: 64 MiB
Trying to boot from MMC1

'''
U-Boot 2020.10 (Oct 12 2020 - 23:08:36 +0000) Allwinner Technology

CPU:   Allwinner V3s (SUN8I 1681)
Model: Lichee Pi Zero
DRAM:  64 MiB
MMC:   mmc@01c0f000: 0
Loading Environment from FAT... Unable to use mmc 0:2... In:    serial@01c28000
Out:   serial@01c28000
Err:   serial@01c28000                                                                                                        
Net:   No ethernet found.                                                                                                     
starting USB...                                                                                                               
No working controllers found                                                                                                  
Hit any key to stop autoboot:  0                                                                                              
switch to partitions #0, OK                                                                                                   
mmc0 is current device                                                                                                        
Scanning mmc 0:2...                                                                                                           
Found U-Boot script /boot/boot.scr                                                                                            
708 bytes read in 2 ms (345.7 KiB/s)                                                                                          
## Executing script at 41900000                                                                                               
4264448 bytes read in 194 ms (21 MiB/s)                                                                                       
efi_free_pool: illegal free 0x42d70040                                                                                        
efi_free_pool: illegal free 0x42d6c040                                                                                        
19744959 bytes read in 890 ms (21.2 MiB/s)                                                                                    
8479 bytes read in 10 ms (827.1 KiB/s)                                                                                        
## Loading init Ramdisk from Legacy Image at 41b00000 ...                                                                     
   Image Name:                                                                                                                
   Image Type:   ARM Linux RAMDisk Image (uncompressed)                                                                       
   Data Size:    19744895 Bytes = 18.8 MiB                                                                                    
   Load Address: 00000000                                                                                                     
   Entry Point:  00000000                                                                                                     
   Verifying Checksum ... Bad Data CRC                                                                                        
Ramdisk image is corrupt or invalid                                                                                           
SCRIPT FAILED: continuing...                                                                                                  
starting USB...                                                                                                               
No working controllers found                                                                                                  
USB is stopped. Please issue 'usb start' first.                                                                               
starting USB...                                                                                                               
No working controllers found                                                                                                  
No ethernet found.                                                                                                            
missing environment variable: pxeuuid                                                                                         
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/00000000                                                                                        
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/0000000                                                                                         
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/000000                                                                                          
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/00000                                                                                           
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/0000                                                                                            
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/000                                                                                             
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/00                                                                                              
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/0                                                                                               
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi                                                                         
No ethernet found.                                                                                                            
missing environment variable: bootfile                                                                                        
Retrieving file: pxelinux.cfg/default-arm-sunxi                                                                               
No ethernet found.                                                                                                            
[... cut ...]
starting USB...                                                                                                               
No working controllers found                                                                                                  
No ethernet found.                                                                                                            
No ethernet found.                                                                                                            
=> 
```

johang, 2020-10-13:
Allwinner V3s has only 64 MB DRAM. I suspect there's some overwrite going on when U-Boot loads the kernel and/or initrd.

More specifically the boot script (https://github.com/johang/sd-card-images/blob/master/2nd-stage-files/pre-2nd-stage-files-armhf/etc/initramfs/post-update.d/zz-update-uimg#L16) load the kernel and initrd (in that order) to the addresses specified by U-Boot (https://github.com/u-boot/u-boot/blob/master/include/configs/sunxi-common.h#L317). You can see the kernel read (4264448 bytes) and initrd read (19744959 bytes) in your log.

ekawahyu, 2020-10-13:
Wow, that's a lot for uncompressed kernel at 16MB (25% of the entire DRAM). How can I debug this? From u-boot?

johang, 2020-10-13:
Yes, this is an U-Boot issue. I'd try to tweak the load addresses in the boot script to see if it's possible to get the kernel and initrd to load without errors. But to be fair, I think 64 MB is too small of Debian. Debian's kernel and initrd is large because it's generic.

peterbmckinley, 2023-01-09:
Hi Johang,

if Debian can't boot on the Lichee Zero why is it STILL listed as a supported board? 

I have seen reference [here](https://blog.jmdawson.co.uk/lichee-nano-pi-will-it-run-debian/) to Debian running successfully on a Lichee Nano (Allwinner F1C100s which only has 32MB RAM). Once booted, the RAM overhead is only 7MB.

Is there any way your image can be tweaked to work? I am assuming you have the hardware

peterbmckinley, 2023-01-09:
Please note for the Bullseye image booting hangs completely at "Starting Kernel"

However for the Bookworm image booting get stuck in a loop resetting at the following point:

## Flattened Device Tree blob at 41800000
   Booting using the fdt blob at 0x41800000
   Loading Ramdisk to 4170c000, end 42dff871 ... OK
ERROR: image is not a fdt - must RESET the board to recover.
FDT creation failed!
resetting ...

U-Boot SPL 2022.04 (Jan 01 2023 - 00:24:15 +0000)

The Sid image as as per the Bookworm image:
ERROR: image is not a fdt - must RESET the board to recover.
FDT creation failed!
resetting ...

Seems to be different issues affecting Bullseye and Bookworm/Sid, if that helps

This repo would be the most fantastic resource if a bit more work was done to make these images work, which I'm confident that they can. Unfortunately I don't personally have the skills to make the adjustments necessary, I can only test.

johang, 2023-01-10:
Dropped support for this board. Not enough memory.

closed 2023-01-10 by johang