johang/sd-card-images #3: Boot image boot-a10s_olinuxino_micro.bin - can't read superblock

stormy-gh, 2021-01-25:
Boot image boot-a10s_olinuxino_micro.bin is damaged.

Steps to reproduce:

$  wget http://ftp.sd-card-images.johang.se/20210125_000001/boot-a10s_olinuxino_micro.bin.gz

$ gunzip boot-a10s_olinuxino_micro.bin.gz

$ fdisk -lu boot-a10s_olinuxino_micro.bin
Dysk boot-a10s_olinuxino_micro.bin: 32 MiB, bajtów: 33554432, sektorów: 65536
Jednostki: sektorów, czyli 1 * 512 = 512 bajtów
Rozmiar sektora (logiczny/fizyczny) w bajtach: 512 / 512
Rozmiar we/wy (minimalny/optymalny) w bajtach: 512 / 512
Typ etykiety dysku: dos
Identyfikator dysku: 0x7e60c54d

Urządzenie                     Rozruch Początek  Koniec Sektory Rozmiar Id Typ
boot-a10s_olinuxino_micro.bin1             2048   65535   63488     31M  c W95 FAT32 (LBA)
boot-a10s_olinuxino_micro.bin2 *          65536 3903487 3837952    1,9G 83 Linux

stormy@darkstar:~$ sudo mount -o loop,offset=1048576 boot-a10s_olinuxino_micro.bin /mnt
mount: /mnt: can't read superblock on /dev/loop4.

The same issue when I write image to SD card:

$ cat /proc/partitions
major minor  #blocks  name
[....]
8       32   15558144 sdc
8       33      31744 sdc1
8       34    1918976 sdc2

$ sudo mount /dev/sdc1 /mnt
mount: /mnt:  can't read superblock on /dev/sdc1.

$ sudo fsck.vfat /dev/sdc1
fsck.fat 4.1 (2017-01-24)
Logical sector size is zero.

johang, 2021-01-25:
Not damaged. The FAT32 partition is not used on Allwinner so it's zeroed. I guess I could format it to avoid confusion.

stormy-gh, 2021-01-26:
Maybe you are right, but I am still unable to boot my olinuxino a10S micro from your image.
It is working with original Olimex image (old Debian stretch).
Normally Olinuxino turn on green LED when begin loading image. 
With your image nothing happens - it means that it can't load image.
Do you have any idea what may cause problem?

johang, 2021-01-26:
Are you sure your image is correct? Just boot-a10s_olinuxino_micro.bin won't work. You need the Debian part too. Total image size should be around 2 GB. If your image is correct you should hook up the serial console and see what it's doing.

stormy-gh, 2021-01-28:
Yes, I am sure. I strictly follow your instruction and sd-card.img is around 2GB.

OlinuXino 10S micro is strange board. It looks that it need to load something from SD card to be able to output to serial console.
I hook up serial console and this is result:

1. Power ON without SD card inserted.
=> Console output: nothing

2. Power ON with SD card with your image A10S
=> Console output: nothing

3. Power ON with SD card with your image A13 (i made this test because Allwinner A10S and A13 are almost the same CPU:
=> Console output: nothing

4. Power ON with original Olimes image (original image is using u-boot):
=> Console output:

U-Boot SPL 2013.10-rc2-08396-g27c7eed-dirty (Oct 04 2013 - 11:56:01)
Board: A10s-OLinuXino-M
DRAM: 512 MiB
CPU: 1008000000Hz, AXI/AHB/APB: 3/2/2
SPL: Please implement spl_start_uboot() for your board
SPL: Direct Linux boot not active!

U-Boot 2013.10-rc2-08396-g27c7eed-dirty (Oct 04 2013 - 11:56:01) Allwinner Technology

CPU:   Allwinner A13/A10s (SUN5I)
Board: A10s-OLinuXino-M
I2C:   ready
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0

In:    serial
Out:   serial
Err:   serial
Net:   emac
Hit any key to stop autoboot:  0 
reading uEnv.txt
** Unable to read file uEnv.txt **
reading boot.scr
** Unable to read file boot.scr **
reading script.bin
35776 bytes read in 5 ms (6.8 MiB/s)
reading uImage
4173440 bytes read in 205 ms (19.4 MiB/s)
## Booting kernel from Legacy Image at 48000000 ...
Image Name:   Linux-3.4.61+
Created:      2013-10-18   6:08:03 UTC
Image Type:   ARM Linux Kernel Image (uncompressed)
Data Size:    4173376 Bytes = 4 MiB
Load Address: 40008000
Entry Point:  40008000
Verifying Checksum ... OK
Loading Kernel Image ... OK

Starting kernel ...

<6>Booting Linux on physical CPU 0
<6>Initializing cgroup subsys cpuset
[...]

I will give up. This is simple device with 7" touchscreen and I managed to install modern firefox on old Debian Wheezy and I can stay with original Olimex image.

johang, 2021-01-28:
Seems like a bootloader issue. Can you share a link to the working image?

johang, 2021-01-28:
I don't own a A10s device, but it works fine on one of my A20 devices.

stormy-gh, 2021-01-28:
Yes, I think so it is issue with bootloader.
Here is the link: ftp://staging.olimex.com/Allwinner_Images/A10S-OLinuXino-MICRO/Debian/sd/

johang, 2021-01-28:
Now I remember that there was another guy that had a similar problem with his Olinuxino A13 board. It turned out to be a SD card issue. See issue #1.

Also you could try to overwrite the bootloader in my image with the bootloader in your image. Looks like it starts at 0x2000 and ends at 0x750d4 in the file. If you write it to the same location in my image (0x2000) it should at least show output in the serial console.

stormy-gh, 2021-02-01:
SD card is 100% because this same card works with Olimex image.

Unfortunately my device is now mounted on the wall and "in production use" so i can't experiment with bootloader now.