johang/sd-card-images #5: rockpro64 bullseye

moserwi, 2022-03-14:
flashed the image to the 32Gb emmc. Boot stuck on.
Loading init Ramdisk....
Verifying checksum ... OK
Flattened Device tree blob at 01f00000
Booting using the fdt blob at 0x1f00000
## last line unreadable... outside screen range.

the emmc is a brand new one.

Thank you for your superb work on the images...

johang, 2022-04-30:
Problem with eMMC only? Can it boot the image from SD card?

moserwi, 2022-05-01:
I'am not sure if I checked that. I had an elder image from elsewhere on another SBC and used it.

moserwi, 2022-05-01:
for me it looked like the installer wanted to boot an image from an tftp server

johang, 2022-05-01:
Can you show the full log?

moserwi, 2022-05-01:
No, I only had a screenshot until the last lne.

>  
>
>   Can you show the full log?

Links:
------
[1] https://github.com/johang/sd-card-images/issues/5#issuecomment-1114257502
[2]  
https://github.com/
Liebe Grüße

Wilhelm Moser
Kontaktdaten:
Wilhelm MOSER

Gentzgasse 65/13
A-1180 WIEN
mobile: +4369919294126
OE1WMR

http://www.moser-willi.at
http://www.net.co.at
http://www.demos.at

ACHTUNG: Diese Email wurde von meinem privaten Mailserver versendet. Ihr
Mailserver übersendet meinem eine MESSAGEQUEUEID wenn er die Nachricht
akzeptiert und gespeichert hat. Diese Bestätigung ist BESSER ALS
"EINSCHREIBEN" da sie den Erhalt genau dieser Email bestätigt.

HINWEIS: Die in dieser Nachricht enthaltenen Informationen, Aussagen
und/oder Erklärungen etc. dienen ausschliesslich der Information des/der
oben bezeichneten Adressaten. Unter keinen Umständen sind in dieser
Nachricht enthaltene Informationen, Aussagen und/oder Erklärungen etc.
rechtlich verbindlich, noch stellen diese eine in irgendeiner Weise
rechtlich beachtliche Handlung des Absenders der Nachricht dar.

PLEASE NOTE: The information contained in this message, including any
statements and/or declarations etc. made therein, serve as information
intended exclusively for the above recipient(s). Under no circumstances is
any of the information, statements and/or declarations made therein legally
binding, nor does it represent any legal obligation on the sender of this
message.