johang/sd-card-images #6: Licheepi Zero hangs at "Starting kernel..."

peterbmckinley, 2022-04-15:
U-Boot 2022.01 (Apr 01 2022 - 03:47:48 +0000) Allwinner Technology

CPU:   Allwinner V3s (SUN8I 1681)
Model: Lichee Pi Zero
DRAM:  64 MiB
WDT:   Not starting watchdog@1c20ca0
MMC:   mmc@1c0f000: 0
Loading Environment from FAT... Unable to use mmc 0:2... In:    serial@1c28000
Out:   serial@1c28000
Err:   serial@1c28000
Net:   No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:2...
Found U-Boot script /boot/boot.scr
899 bytes read in 2 ms (438.5 KiB/s)
## Executing script at 41900000
5001728 bytes read in 211 ms (22.6 MiB/s)
22567854 bytes read in 945 ms (22.8 MiB/s)
11445 bytes read in 54 ms (206.1 KiB/s)
Kernel image @ 0x41000000 [ 0x000000 - 0x4c5200 ]
## Loading init Ramdisk from Legacy Image at 41c00000 ...
   Image Name:
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    22567790 Bytes = 21.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 41800000
   Booting using the fdt blob at 0x41800000
   Loading Ramdisk to 4187a000, end 42dffb6e ... OK
   Loading Device Tree to 41874000, end 41879cb4 ... OK

Starting kernel ...

peterbmckinley, 2022-04-15:
To be clear, I tried a different power supply, rated at 2 Amps, and a reliable Signalex micro USB cable rated at 1.5 Amps. Current draw when boot hangs is a constant 0.16 Amps (160mA), so I can rule out power issues

peterbmckinley, 2022-04-15:
If it helps, I powered up via usb on a host pc, and lsusb gives:

Bus 006 Device 005: ID 1f3a:efe8 Onda (unverified) V972 tablet in flashing mode

johang, 2022-04-30:
Licheepi Zero only has 64 MB RAM. I don't think it can run the standard Debian kernel.

peterbmckinley, 2023-01-09:
Hi Johang,

if Debian can't boot on the Lichee Zero why is it STILL listed as a supported board? 

I have seen reference [here](https://blog.jmdawson.co.uk/lichee-nano-pi-will-it-run-debian/) to Debian running successfully on a Lichee Nano (Allwinner F1C100s which only has 32MB RAM). Once booted, the RAM overhead is only 7MB.

Is there any way your image can be tweaked to work? I am assuming you have the hardware

peterbmckinley, 2023-01-09:
Please note for the Bullseye image booting hangs completely at "Starting Kernel"

However for the Bookworm image booting get stuck in a loop resetting at the following point:

## Flattened Device Tree blob at 41800000
   Booting using the fdt blob at 0x41800000
   Loading Ramdisk to 4170c000, end 42dff871 ... OK
ERROR: image is not a fdt - must RESET the board to recover.
FDT creation failed!
resetting ...

U-Boot SPL 2022.04 (Jan 01 2023 - 00:24:15 +0000)

The Sid image as as per the Bookworm image:
ERROR: image is not a fdt - must RESET the board to recover.
FDT creation failed!
resetting ...

Seems to be different issues affecting Bullseye and Bookworm/Sid, if that helps

This repo would be the most fantastic resource if a bit more work was done to make these images work, which I'm confident that they can. Unfortunately I don't personally have the skills to make the adjustments necessary, I can only test.

johang, 2023-01-10:
Dropped support for this board. Not enough memory.

closed 2023-01-10 by johang