johang/sd-card-images #7: USB does not work on Cubieboard 4

som1712, 2022-06-11:
None of the USB ports work.
[putty.log](https://github.com/johang/sd-card-images/files/8884789/putty.log)

johang, 2022-06-12:
Try Debian Bookworm. It has a newer kernel.

som1712, 2022-06-12:
I did the tests with the Bookworm version, but without success. I don't have the 5V voltage on the USB port. With the image available on the Cubieboard website, it works perfectly.

som1712, 2026-10-31:
## Workaround: USB ports dead on the Debian 13 image (Cubieboard 4)

**Cause.** Two separate things are missing from the device tree handed to the kernel (tested on kernel `6.12.113+deb13-armmp`, U-Boot `2026.07johang`):

1. All USB nodes (`usb@a00000`, `usb@a00400`, `usb@a02000`, `usb@a02400` and the PHYs `phy@a00800` / `phy@a02800`) have `status = "disabled"`, so `ehci-platform` / `ohci-platform` never probe and nothing about USB shows up in `dmesg`.
2. Enabling them is not enough: the board's 5V VBUS supply for the ports is controlled by two GPIOs (**PH14** and **PH15**), and there is no regulator for them in the DT. The controllers and the onboard hub (`05e3:0608`) show up, but devices plugged into the ports are never detected, with no messages at all. This matches what was reported above (no 5V on the ports).

The pins come from patches posted to linux-arm-kernel for this board (https://lists.infradead.org/pipermail/linux-arm-kernel/2015-May/341561.html); both must be enabled for all ports to have power.

**Quick check (optional, no changes to the system):**

```
apt install gpiod usbutils
gpioset -z -c gpiochip1 238=1 239=1   # PH14 and PH15 (lines 238/239 of gpiochip1)
dmesg -w                              # now plug something in
pkill gpioset                         # release the pins afterwards
```

If the device is detected after `gpioset`, this is your problem. (This uses the libgpiod v2 syntax from Debian 13.)

### Permanent fix

I copy the device tree the board is already running (that is the one U-Boot passes, and it is known to boot), enable the USB nodes, add two always-on 5V regulators and tell U-Boot to use the result. Everything is done on the board; backups are kept.

**1. Install the tools**

```
apt install device-tree-compiler
```

**2. Create the modified DTB**

```
cd /boot
F=sun9i-a80-cubieboard4-usb-vbus.dtb
dtc -I fs -O dtb -o $F /sys/firmware/devicetree/base 2>/dev/null

# enable USB controllers and PHYs (host0 and host2)
for n in phy@a00800 usb@a00000 usb@a00400 phy@a02800 usb@a02000 usb@a02400; do
  fdtput -t s $F /soc@20000/$n status okay
done

# add the two VBUS regulators (PH14, PH15)
PIO=$(fdtget -t u $F /soc@20000/pinctrl@6000800 phandle)
for pin in 14 15; do
  n=/usb-vbus-ph$pin
  fdtput -c $F $n
  fdtput -t s $F $n compatible regulator-fixed
  fdtput -t s $F $n regulator-name usb-vbus-ph$pin
  fdtput -t u $F $n regulator-min-microvolt 5000000
  fdtput -t u $F $n regulator-max-microvolt 5000000
  fdtput -t u $F $n gpio $PIO 7 $pin 0
  fdtput -t s $F $n enable-active-high ""
  fdtput -t s $F $n regulator-always-on ""
  fdtput -t s $F $n regulator-boot-on ""
done
```

Check the result (you should see 6 `okay` USB nodes and the two `usb-vbus-ph*` nodes, with `gpio = <... 0x07 0x0e ...>` / `0x0f`):

```
dtc -I dtb -O dts $F 2>/dev/null | grep -B1 -A10 "usb-vbus"
```

**3. Make U-Boot load it**

`/boot/extlinux/extlinux.conf` is generated by `u-boot-update`, so configure the generator instead of editing the file:

```
cp /boot/extlinux/extlinux.conf /root/extlinux.conf.bak
sed -i '/^U_BOOT_FDT=/d' /etc/default/u-boot
echo 'U_BOOT_FDT="/boot/sun9i-a80-cubieboard4-usb-vbus.dtb"' >> /etc/default/u-boot
u-boot-update
grep -nE "fdt|append" /boot/extlinux/extlinux.conf
```

You should see `fdt /boot/sun9i-a80-cubieboard4-usb-vbus.dtb` and `append root=LABEL=rootfs rw rootwait`. Make sure `root=` is still there before rebooting.

**4. Reboot and verify**

```
reboot
# after boot:
lsusb -t
dmesg | grep -iE "ehci|ohci|usb [0-9]"
grep -i vbus /sys/kernel/debug/regulator/regulator_summary
```

Plug in a mouse/flash drive: it should be detected on all ports without running `gpioset`.

### Notes

- The ports are always powered (the regulators are `always-on`).
- The `.dtb` is a copy of the DT that U-Boot provides in this image. If you change the image or update U-Boot, redo step 2.
- If the board does not boot, mount the SD card on another PC and restore `/root/extlinux.conf.bak` to `/boot/extlinux/extlinux.conf` (and remove the `U_BOOT_FDT` line from `/etc/default/u-boot`).
- I did not work out which pin feeds which physical port, which is why both are enabled. I also did not verify the USB 3.0 port, which has no support in mainline for the A80.

It would be nice if the image's U-Boot/device tree carried this fix, since upstream disables USB on this board. @johang, would you consider including it?