johang/sd-card-images #8: A10-Olinuxino-lime

xtrinch, 2022-06-14:
First of all thank you for your hard work,

I've booted into the A10-Olinuxino-lime image, and I'm unable to get apt to work, it says `Temporary failure resolving`. It appears I'm connected and have an IP on eth0 (Can see it in the ifconfig). Can't ping since ping isn't installed.

Can confirm same ethernet port connection setup works with official Olimex images.

What could it be?

evil-publishing, 2022-06-17:
Gotcha. In /etc/resolv.conf the nameserver is set to 192.168.2.1 but in my case it should be 192.168.178.1

```
root@debian:/etc# apt update
Get:1 http://security.debian.org/debian-security bookworm-security InRelease [35.6 kB]
Get:2 http://deb.debian.org/debian bookworm InRelease [130 kB]
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [36.8 kB]
Reading package lists... Done
E: Release file for http://security.debian.org/debian-security/dists/bookworm-security/InRelease is not valid yet (invalid for another 92d 2h 50min 11s). Updates for this repository will not be applied.
E: Release file for http://deb.debian.org/debian/dists/bookworm/InRelease is not valid yet (invalid for another 93d 20h 11min 46s). Updates for this repository will not be applied.
E: Release file for http://deb.debian.org/debian/dists/bookworm-updates/InRelease is not valid yet (invalid for another 93d 20h 11min 45s). Updates for this repository will not be applied.
```
Looks way better - but installing nano and some sort of ping command thru .deb-files and scp wasn't soo nice ;)

Next thing was the time and date was not set correctly and where a little offset. a `date -s `corrected this one.

johang, 2022-06-17:
Nice catch. I made a fix for it.

commit 8e9569ea (2022-06-17): build-debian: Enable systemd-resolved

Fixes #8.
closed 2022-06-17 by johang