johang/sd-card-images #13: ROCK PI 4 PLUS v1.73

jogc, 2022-10-02:
I have a board marked "ROCK PI 4 PLUS v1.73" that I can't get to work. I tried the images for "ROCK Pi 4 model B" and "ROCK Pi 4 model C" (tried both Bullseye and Bookworm) and in all cases I get solid green + blue LED light and no HDMI output. It does boot with rockpi-4b-debian-buster-xfce4-arm64-20220801-0347-gpt.img from Radxa but I would really prefer to run pure Debian on it.

Output from lshw:
https://pastebin.com/fx5Y3i7f

Output from dmesg:
https://pastebin.com/E4uGmvTb

(Maybe this should be seen as a feature request but from what I have found out this board is a variant of ROCK Pi 4B, so I guess these images are supposed to work?)

johang, 2022-10-02:
Connect UART and check console output during boot.

jogc, 2022-10-12:
So after some trial and error realizing 9600 is the correct baud rate with this image and not 1500000 as with Radxa's Debian, I do get output on the console and it does boot. Three issues:

1. Both the blue and green LEDs stay constantly on and they don't do that when I boot with either Radxa's Debian or with Armbian (Armbian_22.08.1_Rockpi-4b_jammy_current_5.15.63.img.xz)
2. Still no output on HDMI. I guess some boot flag is needed?
3. I get this soon after boot, don't know if it's significant or not:

```
[   92.205378] irq 52: nobody cared (try booting with the "irqpoll" option)
[   92.295123] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.0-18-arm64 #1 Debian 5.10.140-1
[   92.405856] Hardware name: ROCK PI 4 Board (DT)
[   92.466474] Call trace:
[   92.499132]  dump_backtrace+0x0/0x1e4
[   92.548097]  show_stack+0x24/0x30
[   92.592399]  dump_stack+0xd0/0x12c
[   92.637867]  __report_bad_irq+0x54/0x180
[   92.690329]  note_interrupt+0x2dc/0x3e0
[   92.741625]  handle_irq_event+0x120/0x1bc
[   92.795248]  handle_level_irq+0xe4/0x21c
[   92.847710]  generic_handle_irq+0x3c/0x54
[   92.901340]  rockchip_irq_demux+0xcc/0x254
[   92.956130]  __handle_domain_irq+0x88/0xf0
[   93.010919]  gic_handle_irq+0xcc/0x14c
[   93.061047]  el1_irq+0xcc/0x180
[   93.103014]  arch_cpu_idle+0x18/0x30
[   93.150817]  default_idle_call+0x4c/0x180
[   93.204443]  do_idle+0x238/0x2b0
[   93.224958] rk3x-i2c ff3c0000.i2c: timeout, ipd: 0x10, state: 1
[   93.247583]  cpu_startup_entry+0x30/0xa0
[   93.326943] rk808 0-001b: Failed to read IRQ status: -110
[   93.379313]  rest_init+0xcc/0xdc
[   93.379326]  arch_call_rest_init+0x1c/0x28
[   93.379334]  start_kernel+0x560/0x598
[   93.379341] handlers:
[   93.379353] [<0000000025cc9704>] irq_default_primary_handler threaded [<00000000228bb210>] regmap_irq_thread
[   93.379370] Disabling IRQ #52
```

johang, 2022-10-14:
These images are very minimal. HDMI out is not guaranteed to work out of the box.

sakittelsen, 2025-01-06:
Only the HDMI output #2 (the one closest to the audio output connector) actually works. The main HDMI output (closest to the power plug) is always dead on these images.