johang/sd-card-images #19: Empty /boot partition, kernel and u-boot files in rootfs

chschnell, 2022-10-19:
First off, thanks a lot for this impressive work.

I tried to build an image for Banana Pi BPI-M2-BERRY with Bullseye using:

    sudo docker run --rm -v /tmp/sd-images:/artifacts sd-images build-boot \
        banana_pi_m2_berry allwinner-v40 bananapi_m2_berry_defconfig arm-linux-gnueabihf

    sudo docker run --rm -v /tmp/sd-images:/artifacts sd-images build-debian \
        debian armhf bullseye

Two binary files were created, as expected, I concatenated them using:

    zcat /tmp/sd-images/boot-banana_pi_m2_berry.bin.gz \
         /tmp/sd-images/debian-bullseye-armhf-mo3oop.bin.gz > sd-card.img
     
After flashing 'sd-card.img` to my SD card the device didn't boot. When inspecting the file systems of both .bin files I noticed that:

1. /boot partition in `boot-...bin.gz` is empty
2. /boot directory in `debian-...bin.gz` contains kernel and u-boot files
3. trying to move the files from `debian-...bin.gz` to `boot-...bin.gz` failed due to lack of space in the 32M /boot partition

I then tried the same for a Raspberry Pi Zero 2 W and the final image turned out to be structurally ok (I didn't boot it though): two partitions (vfat and ext4), boot files in /boot and /boot mount point in rootfs empty.

Finally I ran another attempt for a Banana Pi BPI-M2-ZERO, but it failed for the same problem my initial attempt (Banana Pi BPI-M2-BERRY) failed, vfat /boot partition is empty.

I also checked your downloadable binaries for the two Banana Pis, same problem (mount them and check yourself).

johang, 2022-10-19:
Allwinner images have empty FAT partition by design. The bootloader is written to a space between partition table and the FAT partition. The bootloader then loads boot.scr from /boot in the ext4 partition, which then loads dtb and kernel from the ext4 partition.

Connect to the UART and show what is outputs during boot.

chschnell, 2022-10-19:
Thanks for the quick clarification, I knew about the bootloader but it's the first time I come across this file layout.

This time it booted perfectly fine, I was simply too impatient earlier (my monitor didn't get a signal for some time). Found that out after connecting the serial port.

I very much apologize for my blunder, and again thanks a lot for your effort here, love it!

closed 2022-10-19 by johang