johang/sd-card-images #29: Sync time on startup with ntpdate

deftdawg, 2022-12-12:
First, thanks for making these images, I'm trying to use it to build an Octoprint server on an old PCDuino3 nano.

It boots perfectly and seems to run reasonably well, but it seems NTP does come up on start and sync the time.   The lack of synced time creates issues with SSL certificate validations which impacts the ability to download packages including `ntpdate`.

I can manually set the date by copying it from another machine
```sh
# Time Synced machine
date +%s # copy time from a synced machine

# On the PCduino - set the time to something close
date -s "$(date -d @1670738614)"
```

After `ntpdate` is installed, I can manually sync time by using `ntpdate tick.utoronto.ca` , obviously it would be far better if the images automatically synced their time.

johang, 2022-12-12:
By default it uses NTP servers returned by the DHCP server. You need to either configure your DHCP server or configure Debian.

closed 2022-12-12 by johang

deftdawg, 2022-12-12:
Reading up on it a bit more, I guess this is possibly supposed to be handled by systemd's `systemd-timesyncd.service` unit... 

Looks like it's not working for me, maybe the lack of `/etc/systemd/timesyncd.conf` is the issue?

```
root@octo-et4:/etc/systemd# timedatectl show-timesync --all
Failed to parse bus message: No route to host
root@octo-et4:/etc/systemd# timedatectl status
               Local time: Tue 2022-12-06 15:32:18 EST
           Universal time: Tue 2022-12-06 20:32:18 UTC
                 RTC time: Tue 1965-07-20 06:36:28
                Time zone: America/Toronto (EST, -0500)
System clock synchronized: no
              NTP service: n/a
          RTC in local TZ: no
root@octo-et4:/etc/systemd# timedatectl show  
Timezone=America/Toronto
LocalRTC=no
CanNTP=no
NTP=no
NTPSynchronized=no
TimeUSec=Tue 2022-12-06 15:32:26 EST
RTCTimeUSec=--- XXXX-XX-XX XX:XX:XX
```

deftdawg, 2022-12-12:
systemd-timesyncd wasn't installed by default, adding `systemd-timesyncd` fixed it instantly:
```sh
apt install systemd-timesyncd
```

peterbmckinley, 2023-01-09:
Please note you cannot install or update anything, including systemd-timesyncd, on any of these images until you manually bring system time close to current after installation using the following command (example for just now, modify as required):
```
date --set="9 JAN 2023 22:48:00"
```

commit 0a086589 (2023-03-05): build-debian: Install systemd-timesyncd

Fixes #29.