johang/sd-card-images #38: Nameserver incorrect on Debian Bookworm for Odroid N2+

rodw-au, 2023-02-14:
#8 refers - data in resolv.conf is still wrong in Bookworm on Odroid N2+. (same as N2 just with faster processor)
I know the N2+ is on the network because ifconfig reports an IP address and I can ping that IP from another PC on the network.

I can diagnose the fault but I don't know how to fix this in the absence of a text editor.
PS I am really excited to find your images.... Now I just have to get the networking working...

Any help fixing this would be appreciated.

johang, 2023-02-14:
> /etc/resolve.conf contains nameserver 127.0.0.53 (which is nonsense)

This is not nonsense. It's pointing to systemd-resolved.

>  It would be helpful if the following debian packages were included so problems like this could be resolved:

You can install all of these with apt.

rodw-au, 2023-02-15:
Maybe I assumed to quickly it was related to Issue #8 so I apologise for not understanding how resolve works.

Let me  rephrase the issue. **I am unable to install anything with apt.**
I do use apt quite a lot so do know how it works.

I tried several times with several installs of all the images for the odroid N2 you provide.
(Only sid and bookworm boot to a logon screen but I'll raise the issues I see when I get past this hurdle.)

Its certainly an issue as I cannot install anything from the repos.
I checked sources.list is set correctly for debian
I see downloads on an apt update 
resolvectl status shows 192.168.1.1 which is my gateway.
apt can't find anything that is in the repos
I'm not sure how to proceed or troubleshoot.

johang, 2023-02-15:
Any logs or error messages?

rodw-au, 2023-02-15:
I'll have another go when I get a bit of time and take some photos of the screen.

**apt update** seems to run correctly but says something about you can't use something yet.

**cat /etc/apt/sources.list** seems to show a normally formatted file but includes security settings. 
I don't believe security repos should be included while Bookworm is still Tessting branch so I assumed that was what generated the warning but did not expect that to be an issue.

**apt upgrade** says there is nothing to upgrade, no errors

**apt install some-package** says the package is not found despite being in the debian repos

**apt install nano says nano** is not in the repository but is mentioned  elsewhere. The only on e I tried which returned this message.

rodw-au, 2023-02-16:
Ok I sorted this and updated the title accordingly.
The System date and  time was set to 30 Jan (18 days ago)
This throws the error
`Release file for http://deb.debian/dists/bookworm/InRelease is not valid yet`
Solution is to:
`
# Refer https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
timedatectl set-timezone 'Australia/Brisbane' 
timedatectl set-time '2023-02-16 17;31'  #Current local time
timedatectl  status   #to confirm settings
apt update
apt upgrade
`

Sorry for being so off the mark initially. I had never experienced this issue before.
Perhaps this could be documented somewhere.

closed 2023-02-16 by rodw-au

johang, 2023-02-16:
Could be that your NTP client doesn't have any NTP server to connect to. I pushed a change that adds fallback NTP servers in case the DHCP server is not providing any.