johang/sd-card-images #39: Reboot does not work.

NEMSLinux, 2023-02-17:
Hello.
Upon initiating a reboot via `/usr/sbin/reboot` or `/usr/sbin/shutdown -r now` the system halts, but does not reboot. A hard power cycle is required to get back up and running.

I've tested the Khadas VIM3 image as well as the ODROID N2 image, and both exhibit this behavior.

I tried reinstalling `systemd-sysv` just for good measure, but no success.

Any chance others are seeing this behavior?

Thanks for the great project!!

NEMSLinux, 2023-02-17:
**Update:** I tested the ODROID N2 with Bookworm (instead of Bullseye) and the issue does not exist. So perhaps this is a problem with the Bullseye version only (I would rather use Bullseye at this time).

joshuaboniface, 2023-06-06:
Confirming this is happening with a AML-S905X-CC as well. `reboot`, `shutdown -h`, etc. do not work properly.

johang, 2023-06-06:
Most likely a kernel issue in something AML specific.

arkadijs, 2023-08-02:
No reboot on Odroid C2 either. A problem with SD peripheral left in wrong state? Boots fine after power cycle. 
Boot process looping with the following messages on serial console:

```
BL2 Built : 11:44:26, Nov 25 2015. 
gxb gfb13a3b-c2 - jcao@wonton

Board ID = 8
set vcck to 1100 mv
set vddee to 1050 mv
CPU clk: 1536MHz
DDR channel setting: DDR0 Rank0+1 same
DDR0: 2048MB(auto) @ 912MHz(2T)-13
DataBus test pass!
AddrBus test pass!
Load fip header from SD, src: 0x0000c200, des: 0x01400000, size: 0x000000b0
Load bl30 from SD, src: 0x00010200, des: 0x01000000, size: 0x00009ef0
sd/emmc read data error: ret=16
aml log : SIG CHK : 351 for address 0x01000000
reset...
GXBB:BL1:08dafd:0a8993;FEAT:EDFC318C;POC:3;RCY:0;EMMC:800;NAND:81;SD:0;READ:800;USB:8;LOOP:1;EMMC:800;NAND:81;SD:0;READ;
TE: 5141139
no sdio debug board detected 
```