johang/sd-card-images #43: ROC-RK3328-CC Images No Display

petcat30, 2023-02-26:
Hello, I have tried both Bookworm and Bullseye images for the ROC-RK3328-CC and I'm not able to get my display to work. The keyboard lights up and I see the Ethernet port blinking. The monitor is an Acer GN246HL hooked up w/ HDMI.

johang, 2023-02-28:
These images are very minimal. You shouldn't expect display to work out of the box.

petcat30, 2023-03-02:
Ah I see, I'm trying an image again. It's been about 15 minutes since I powered on the board and it isn't showing up on my router device list. Any thoughts?

johang, 2023-03-13:
Hard to say. Can you show the output on the UART console during boot?

petcat30, 2023-03-19:
I think my issue is I assumed the Firefly roc-rk3328-cc image would work with a Libre Computer Renegade because there isn't a Renegade image. I don't have a UART anyways so gonna close this.

closed 2023-03-19 by petcat30