johang/sd-card-images #49: Libre ROC-RK3328-CC not booting

wilbert-vb, 2023-04-23:
I created sd-images for all three available Debian releases following the instructions (cat a b > c)
None of them boot.
The Libre board shows solid yellow and red LEDS (no heartbeat), no HDMI output and network is offline.

The sd-images look alright, FAT32 partition and linux partition, the linux filesystem shows no errors.
The FAT partition is not formatted.

The Libre board boots Raspbian and Armbian with success from the same microSD cards.

How can I help solve this?

johang, 2023-04-23:
Please connect to the UART console and show the output during boot.

wilbert-vb, 2023-05-10:
Johan, it took me this long to find an adapter that supports 1.5Mps.

Anyway, I was able to generate the attached log file.
[screenlog.0.zip](https://github.com/johang/sd-card-images/files/11446558/screenlog.0.zip)

Let me know how I can help.

wilbert-vb, 2023-05-19:
> ALERT!  PARTUUID=d04916bb-02 does not exist.

blkid output shows:
`PARTUUID="d04916bb-01"`

johang, 2023-05-19:
U-boot clearly loads the kernel from SD card (0xff500000), but looks like Linux then tries to mount the rootfs from the eMMC (0xff520000).

wilbert-vb, 2023-05-19:
> U-boot clearly loads the kernel from SD card (0xff500000), but looks like Linux then tries to mount the rootfs from the eMMC (0xff520000).

The kernel does not find the SD card (dwmmc_rockchip ff500000.mmc)

johang, 2023-05-20:
I wonder why...

@wilbert-vb, can you do cat /proc/modules when it reaches "Gave up waiting for root file system device"?

wilbert-vb, 2023-05-20:
> I wonder why...
> 
> @wilbert-vb, can you do cat /proc/modules when it reaches "Gave up waiting for root file system device"?
[screenlog.1.zip](https://github.com/johang/sd-card-images/files/11521213/screenlog.1.zip)

johang, 2023-05-20:
I think this is a Debian kernel problem and might be fixed by this patch: https://salsa.debian.org/kernel-team/linux/-/commit/1c12e30895c359c2ac27a5fe09fa4d0e665d4525

johang, 2023-05-21:
I added Debian experimental to the builds. It should show up in a few days. You can try to see if it can find the SD-card, but I don't expect the image to work well other than that.

wilbert-vb, 2023-05-21:
I installed a linux-image from the Libre Computer repo
/etc/apt/sources.list.d/libre-computer-deb.list:
`
deb [signed-by=/usr/share/keyrings/libre-computer-deb.gpg] https://deb.libre.computer/repo linux main non-free
`

I also modified u-boot by enabling setexpr:

```
#
# Shell scripting commands
#
# CONFIG_CMD_CAT is not set
CONFIG_CMD_ECHO=y
CONFIG_CMD_ITEST=y
CONFIG_CMD_SOURCE=y
CONFIG_CMD_SETEXPR=y
CONFIG_CMD_SETEXPR_FMT=y

```
And the result is a login prompt.
It looks like I don't have network, though, only console login.

closed 2023-06-06 by johang