johang/sd-card-images #50: Can't upgrade kernel [Bookworm, La Frite)

38github, 2023-04-26:
I have your Debian Bookworm version on a La Frite and I can upgrade packages except a few, most notably the kernel. I always get a message about no space left on device even though everything else can be installed.

```
Setting up libc6-dev:arm64 (2.36-9) ...
Processing triggers for libc-bin (2.36-9) ...
Processing triggers for initramfs-tools (0.142) ...
update-initramfs: Generating /boot/initrd.img-6.1.0-6-arm64
cpio: write error: No space left on device
E: mkinitramfs failure cpio 2
update-initramfs: failed for /boot/initrd.img-6.1.0-6-arm64 with 1.
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 linux-image-6.1.0-7-arm64
 linux-image-arm64
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

johang, 2023-04-26:
Probably not space left on the root filesystem. It's described on the tips and tricks page.

38github, 2023-04-26:
Jag har utökat den och det ska vara ett antal GiB ledigt.

38github, 2023-05-02:
Har du någon aning om vad det kan vara då rotfilsystemet inte är fullt?

38github, 2023-06-04:
Jag verkar ha löst det. Jag ändrade `MODULES=most` till `MODULES=dep` i `/etc/initramfs-tools/initramfs.conf`. Kanske ställer 512MB RAM till det på något vis när cpio ska packa samman init-filen?

johang, 2023-06-04:
OK, nice finding. Looks like mkinitramfs uses `/var/tmp`.

commit 4e86af5f (2023-06-04): Don't mount /var/tmp as tmpfs

mkinitramfs seems to expect a rather large /var/tmp.

Fixes #50.
closed 2023-06-04 by johang