johang/sd-card-images #56: Bookworm(stable) not bootable, SID works (Orange Pi Zero Plus, and maybe more). Infos+Workaround

pm-pm, 2023-06-19:
Hello,

the bookworm image is not bootable, the SID image works. Could you please check it. :-)

Maybe i found the problem, description following. On the other hand you can bring "bookworm" to run with a workaround. See information below.

I only testet "Orange Pi Zero Plus" from here:
https://sd-card-images.johang.se/boards/orange_pi_zero_plus.html

If you follow the Instructions (Download and zcat) the bookworm Image starts, the Ethernet LED are flashing, but no IP is assigned via DHCP. (SID Image in this way works). The bootprocess hangs! I have no serial Console. (The problem is not the image itself, but the creation of the SD Card)

I think the "problem" is the different image size between "Bookworm" and "SID" and maybe the "bookworm" Images does not fit into the created partition!?

**To reproduce the error**, you do not need an Orange PI Zero Plus, because the filesystem on SD Card looks damaged:

1. Download and zcat **bookworm** from instructions, write to SD Card. Keep SD Card in PC. Try to mount the Main partion in your PC. Filesystem access **NOT possible**. (gparted also find errors in Main SD Card Partition)

2. Download and zcat **SID** from instructions, write to SD Card. Keep SD Card in PC. Try to mount the Main partion in your PC. It **works**. You can access the filesystem. 

Conclusion: The concatenation of "bookworm" image is faulty.

.

My personal workaround to make "bookworm" run is copy the Main Partition manually:

```
1.			# Extract files
-rw-rw-r--  1 user user   33554432 Jun 18 14:02 boot-orange_pi_zero_plus.bin
-rw-rw-r--  1 user user  256405633 Jun 18 14:02 debian-bookworm-arm64-oojoh7.bin.gz
-rw-rw-r--  1 user user 3800000000 Jun 18 20:18 sd-card.img

2.			# Copy ONLY Bootimage to SD Card
			# Then you have a valid Partition Table. Second Partition is visible now, but is empty. (Of course)

dd if=/boot-orange_pi_zero_plus.bin /dev/sdX

3.			# Format second (ca. 3.5GB) Partition with ext4 and Attribut "boot"
			# I used "gparted". Follow the instruction. Commandline will also work.

Something like: "mkfs.ext4 /dev/sdX2" # Not tested

4.			# Mount bookworm Image (downloaded) on loop Device

fdisk -lu sd-card.img		# bookwork image (Extracted debian-bookworm-arm64-oojoh7.bin.gz)

Gerät                 Boot Anfang    Ende Sektoren Größe Kn Typ
sd-card.img1        8192   65535    57344   28M  c W95 FAT32 (LBA)
sd-card.img2 *     65536 7419903  7354368  3,5G 83 Linux

Calculate 65536*512 = 33554432

losetup -f		# Find first free loop device XX (Details see link below)

losetup -o 33554432 /dev/loopXX sd-card.img

fsck -fv /dev/loopXX	# If you like, you can file check. I never did it.

mkdir /mnt/bookworm-image

mount /dev/loopXX /mnt/bookworm-image	# Mount extracted bookworm Image File into Filesystem

5.			# Mount the second SD Card Partition. The empty created Partition.

mkdir /mnt/sd-card-partiton
mount /dev/sdX2 /mnt/sd-card-partiton

			# Now both (Downloaded Image and SD Card) Partitionen are mounted in Filesystem. You can copy.

6.			# Copy Data

cp -p -R /mnt/bookworm-image/. /mnt/sd-card-partiton

7. Umount and Eject. Use Card in Orange Pi Zero Plus. - It boots and works.

```

Add on, help from this sites:
https://askubuntu.com/questions/69363/mount-single-partition-from-image-of-entire-disk-device # Mount single partition from image of entire disk (device)
+
https://debianforum.de/forum/viewtopic.php?t=182516 # (gelöst) losetup und partitionierte Image-Datei

pm-pm, 2023-06-19:
Maybe a hint that the filesystem from "bookworm" image does not fit:

```
e2fsck -f -y -v -C 0 /dev/sdc2
e2fsck 1.46.5 (30-Dec-2021)
Die Größe des Dateisystems (laut Superblock) ist **919542** Blöcke.
Die physikalische Größe des Gerätes ist **919296** Blöcke.
Entweder ist der Superblock oder die Partionstabelle beschädigt!
Abbrechen? ja
```

WiseLord, 2023-06-19:
I faced this problem on bananapi-m2-zero yesterday. I fixed it by something like `resize2fs /dev/mmcblk2` (on my laptop with SD-card inserted) and didn't pay much attention to this issue anymore.

pm-pm, 2023-06-19:
Thanks for letting us know about the alternative. Great.

In my case i had to use "resize2fs /dev/sdX2"

kiko81, 2023-06-20:
Confirm same with Bpi m2 berry image - see # 57

sid works as intended - stable - with the suggested workaround

commit 13df092f (2023-06-25): Fix partition sizes

Fixes #56, #58.
closed 2023-06-25 by johang