johang/sd-card-images #58: Bpi m2 Berry doesn't boot

kiko81, 2023-06-19:
### Discussed in https://github.com/johang/sd-card-images/discussions/57

<div type='discussions-op-text'>

<sup>Originally posted by **kiko81** June 19, 2023</sup>
i'm doing exactly as [this instruction](https://sd-card-images.johang.se/boards/banana_pi_m2_berry.html) states - no success - the board doesn't boot. please help
Do i have to point drive or partition when `dd if=sd-card.img of=/dev/sdXXX` ?
I'm pointing drive..</div>

WiseLord, 2023-06-19:
Isn't it related to #57 ?

pm-pm, 2023-06-19:
> ### Discussed in #57
> 
> Originally posted by **kiko81** June 19, 2023 i'm doing exactly as [this instruction](https://sd-card-images.johang.se/boards/banana_pi_m2_berry.html) states - no success - the board doesn't boot. please help Do i have to point drive or partition when `dd if=sd-card.img of=/dev/sdXXX` ? I'm pointing empty drive, respectfully not mounted

You are right. You have to point to DRIVE like /dev/sdb.

kiko81, 2023-06-20:
managed to do it just as #56 workaround - so stable distro concatenation somehow wrong

kiko81, 2023-06-20:
another problem appeared can't ssh to stable
root login is permitted by default in image
made the following changes to `/etc/systemd/network/eth.network`

```
[Match]
Name=eth0

[Network]
IPv6AcceptRA=no

Address=172.17.33.112/24
Gateway=172.17.33.1

DNS=8.8.8.8

[DHCP]
UseDNS=false
```
apt update ok - installed both ping and nano - everything ok
pinging both image and host and 8.8.8.8
but can't ssh to 172.17.33.112 - "Network error: Software caused connection abort"
tried other adersses as well to avoid eventual conflicts - no success
ssh-ing via powershell - no problems with other images

closed 2023-06-20 by kiko81