johang/sd-card-images #59: How to activate/enable Orange Pi (Zero Plus) Extension Board (USB2/USB3/IR)?

pm-pm, 2023-06-19:
Hello,
i have "bookworm" running on my Orange Pi Zero Plus. Fine. :-)

But the Extension does not work. The USB Ports are not working. (Only 5V Power)
I learned to add an "Overlay". Like this:
https://forum.armbian.com/topic/3718-orange-pi-zero-dev-kernel-does-not-see-expansion-usb-ports/

But how to do it **in this Distribution**?
/boot/config.txt?
Adding a *.dtb File?
Where to add?
How to make?

The "sun50i-h5-orangepi-zero-plus.dts" is available, but how to handle/install it?

Thanks.

johang, 2023-06-20:
These images doesn't support overlays right now.

pm-pm, 2023-06-21:
Hello johang,
thanks for reply.
Is it possible to activate the Expansion Board by default? I think most Orange Pi Zeros (Plus) will habe the Expansion Board.
Without the Board you only have 1 USB Port and no IR/Audio.