johang/sd-card-images #64: Odroid C2 Ubuntu images stuck on Starting kernel...

arkadijs, 2023-07-09:
Odroid C2 successfully boots with Debian `bookworm` image but Ubuntu `jammy`, `lunar`, and `mantic` images stops at `Starting kernel...`. Serial console show no progress. Tried SanDisk Ultra and Samsung EVO SDHC cards. The only difference is `Uncompressing Kernel Image` in U-Boot output with Ubuntu images.

Debian:
```
U-Boot 2023.04 (Jul 02 2023 - 23:21:30 +0000) odroid-c2

Model: Hardkernel ODROID-C2
SoC:   Amlogic Meson GXBB (S905) Revision 1f:b (0:1)
DRAM:  2 GiB
Core:  164 devices, 27 uclasses, devicetree: separate
MMC:   mmc@72000: 0, mmc@74000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@c9410000
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:2...
Found U-Boot script /boot/boot.scr
987 bytes read in 3 ms (321.3 KiB/s)
## Executing script at 08000000
32350144 bytes read in 1416 ms (21.8 MiB/s)
35021950 bytes read in 1579 ms (21.2 MiB/s)
Unknown command 'setexpr' - try 'help'
Unknown command 'setexpr' - try 'help'
28168 bytes read in 10 ms (2.7 MiB/s)
Moving Image from 0x8080000 to 0x8200000, end=a180000
## Loading init Ramdisk from Legacy Image at 13000000 ...
   Image Name:   
   Image Type:   AArch64 Linux RAMDisk Image (uncompressed)
   Data Size:    35021886 Bytes = 33.4 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 08008000
   Booting using the fdt blob at 0x8008000
Working FDT set to 8008000
   Loading Ramdisk to 78dd1000, end 7af3743e ... OK
   Loading Device Tree to 0000000078dc7000, end 0000000078dd0e07 ... OK
Working FDT set to 78dc7000

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.0-9-arm64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld ()
[    0.000000] Machine model: Hardkernel ODROID-C2
...
```

Ubuntu:
```
U-Boot 2023.04 (Jul 02 2023 - 23:21:30 +0000) odroid-c2

Model: Hardkernel ODROID-C2
SoC:   Amlogic Meson GXBB (S905) Revision 1f:b (0:1)
DRAM:  2 GiB
Core:  164 devices, 27 uclasses, devicetree: separate
MMC:   mmc@72000: 0, mmc@74000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@c9410000
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:2...
Found U-Boot script /boot/boot.scr
999 bytes read in 2 ms (487.3 KiB/s)
## Executing script at 08000000
16828248 bytes read in 719 ms (22.3 MiB/s)
66363829 bytes read in 2828 ms (22.4 MiB/s)
Unknown command 'setexpr' - try 'help'
Unknown command 'setexpr' - try 'help'
28368 bytes read in 7 ms (3.9 MiB/s)
   Uncompressing Kernel Image
Moving Image from 0x8080000 to 0x8200000, end=b680000
## Loading init Ramdisk from Legacy Image at 13000000 ...
   Image Name:   
   Image Type:   AArch64 Linux RAMDisk Image (uncompressed)
   Data Size:    66363765 Bytes = 63.3 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 08008000
   Booting using the fdt blob at 0x8008000
Working FDT set to 8008000
   Loading Ramdisk to 76ff0000, end 7af3a175 ... OK
   Loading Device Tree to 0000000076fe6000, end 0000000076fefecf ... OK
Working FDT set to 76fe6000

Starting kernel ...
```

johang, 2023-07-13:
Works on my Amlogic board (AML-S905X-CC (Le Potato)). Could be some console issue with that particular board configuration somehow. Check if you can SSH to it.

arkadijs, 2023-07-13:
No luck there. Network not responding, no any / DHCP packets observed by tcpdump. The blue status led is off yet with kernel booted it should be blinking (and it does with Debian).

HDMI and serial are consistent: `Starting kernel ...` is the last message on both.

johang, 2023-07-25:
I will push a patch to reduce Ubuntu's initramfs size. It will be included in the builds on Monday. Try it.

arkadijs, 2023-08-02:
Thank you, this is resolved now - the board is booting into Ubuntu.

closed 2023-08-02 by arkadijs