johang/sd-card-images #81: Raspberry Pi 4B does not boot Debian Testing

knuxyl, 2023-09-19:
I have installed to an sd card and a USB drive and neither will boot. Both boot to the rainbow screen and will not continue. Power supply is more than adequate, used different storage medias, tried both hdmi ports, etc. I cannot provide logs because no logs are being generated. I don't even know if it is loading anything from the media, I'm not too familiar with rpi boot process.

bj-github, 2023-09-21:
I have also encountered this when trying the images in recent months, and not known how to report it. I've usually reverted to using the Debian-provided [Tested Images](https://raspi.debian.net/tested-images/) as a fallback.

knuxyl, 2023-09-21:
> Do you have any logs?

No, it does not generate any logs because the kernel is not loading. I
checked and uboot isnt generating any logs either.

bj-github, 2023-09-23:
Looking through my notes, I discovered my first experience with this was back in May. As noted in the original report, it seems to simply freeze at the RPi "rainbow square" screen.

Kiwimannen, 2024-07-30:
Im still having this problem with rpi4 - only rainbow screen.

Since its the boot file it must effect all the other files as well.

Perhaps this is a noob question but I have tried the rpi4 uefi files from pftf which works, but I’m having trouble booting Debian os. 

I found this when I searched after a complete Debian vanilla image for raspberry pi - is this correct ?

knuxyl, 2024-10-17:
I have just tried latest debian testing again on an sdcard and it still won't boot. I get a rainbow screen. I updated the eeprom to 2013 01 and still doesn't work. UART isn't putting anything out either.

bj-github, 2025-02-28:
Fwiw, the Trixie RPi3B+ image from 2025-02-24 has the same issue: RPi3B+ boot hangs at the "rainbow" screen, with no further information available. In case this "data point" suggesting a broader issue helps.

closed 2026-05-17 by johang