johang/sd-card-images #99: debian 12.4 for MK802 video issue

paoluccigiorgio, 2024-01-02:
Hello, I have tried successfully to install your debian-bookworm-armhf-aevoo9.bin.gz image on my MK802 stick. Everything works flawlessy, wifi is performant (after installing  wireless firmware), overall system stability is excellent. Unfortunately, video is unstable when connected to a monitor through the hdmi connector. Actually video is stable when no cpu activity is performed, but it becomes  deformed when launching commands or doing activities which involve a significant CPU load. The issue is less noticeable in console mode, but really annoying using X where just moving the mouse makes the screen glitch. Probably some video buffer kernel bug.....Any suggestion?

Here is my  uname -a:
Linux debian 6.1.0-16-armmp #1 SMP Debian 6.1.67-1 (2023-12-12) armv7l GNU/Linux

Thank you in advance.
Best regards
Giorgio Paolucci

johang, 2024-01-04:
These images are very minimal. I don't think you should expect advanced features to work well. But regarding your problem, it sounds like it's running in some software rendering and/or framebuffer mode.

paoluccigiorgio, 2024-01-05:
Hi, thankyou anyway.
Bye
Giorgio