johang/sd-card-images #100: Rock 3A, plus others, are missing just about everything to have a functioning device

knuxyl, 2024-01-04:
I've tried booting these images on several boards. I managed to successfully boot only 2, the Rock 3A and I believe the Rock 4B+. Both of these did not include a dhcp client, parted, fdisk, iproute2, or really anything to make it usable. Even if ethernet worked, there is no dhcp client installed to get outside access to install all the missing packages.
The website says to reclaim the lost partition space by using parted, however that is not included so is impossible. This has happened on multiple boards for me, so I'm assuming there is something either broken in the script to include these packages, or a decision was made to remove packages that make these boards unreasonably crippled.
Also I believe it would make more sense to resize these output images to 1-2GB, preferably 1. Most of these boards use very slow sd cards or emmc and having to write 4GB at a time to troubleshoot why somethings isn't working, which is just about inevitable with these arm boards, is very time consuming and I believe unnessecary. 
And thanks for adding the RK2568 and RK3568 boards! I've been checking the website for months waiting for the Rock 3A support.

paoluccigiorgio, 2024-01-04:
Hi, I understand your disappointment, but if you have installed one of Johang's images on a sd card, you can simply download all the .deb packages you need from another linux machine,  then mount the 2nd partition of the sd card and copy all the packages on it. Umount the sd card and boot it on your arm device and then install with dpgs --install.
I  had to install several .debs and wifi firmware as well  to have a networked working system so that i could directly  install all the other pagkages  with apt.
Bests.
Giorgio

johang, 2024-01-04:
The images are very minimal on purpose. But there is a DHCP client (systemd-networkd).

paoluccigiorgio, 2024-01-04:
Hi Knuxyl,
if you want, here are the debs I downloaded to be able to set up network (with dependencies)....Well, may be some more...
Bests.
Giorgio
/root/DEBS/libjs-underscore_1.13.4~dfsg+~1.11.4-3_all.deb
/root/DEBS/libreadline8_8.2-1.3_armhf.deb
/root/DEBS/libpython3-stdlib_3.11.2-1+b1_armhf.deb
/root/DEBS/libctf-nobfd0_2.40-2_armhf.deb
/root/DEBS/openssl_3.0.11-1~deb12u2_armhf.deb
/root/DEBS/iproute2-doc_6.1.0-3_all.deb
/root/DEBS/libpython3.11-minimal_3.11.2-6_armhf.deb
/root/DEBS/libatm1_2.5.1-4+b2_armhf.deb
/root/DEBS/python3_3.11.2-1+b1_armhf.deb
/root/DEBS/libmnl0_1.0.4-3_armhf.deb
/root/DEBS/libjansson4_2.14-2_armhf.deb
/root/DEBS/media-types_10.0.0_all.deb
/root/DEBS/libpcsclite1_1.9.9-2_armhf.deb
/root/DEBS/libffi8_3.4.4-1_armhf.deb
/root/DEBS/isc-dhcp-client-ddns_4.4.3-P1-2_armhf.deb
/root/DEBS/libcap2-bin_2.66-4_armhf.deb
/root/DEBS/libnl-3-200_3.7.0-0.2+b1_armhf.deb
/root/DEBS/libsqlite3-0_3.40.1-2_armhf.deb
/root/DEBS/binutils-common_2.40-2_armhf.deb
/root/DEBS/libxtables12_1.8.9-2_armhf.deb
/root/DEBS/libnl-genl-3-200_3.7.0-0.2+b1_armhf.deb
/root/DEBS/python3.11-minimal_3.11.2-6_armhf.deb
/root/DEBS/python3-minimal_3.11.2-1+b1_armhf.deb
/root/DEBS/libbpf1_1.1.0-1_armhf.deb
/root/DEBS/binutils-arm-linux-gnueabihf_2.40-2_armhf.deb
/root/DEBS/python3.11_3.11.2-6_armhf.deb
/root/DEBS/wpasupplicant_2.10-12_armhf.deb
/root/DEBS/readline-common_8.2-1.3_all.deb
/root/DEBS/libunwind8_1.6.2-3_armhf.deb
/root/DEBS/ca-certificates_20230311_all.deb
/root/DEBS/libatomic1_12.2.0-14_armhf.deb
/root/DEBS/libelf1_0.188-2.1_armhf.deb
/root/DEBS/strace_6.1-0.1_armhf.deb
/root/DEBS/libjs-jquery_3.6.1+dfsg+~3.5.14-1_all.deb
/root/DEBS/isc-dhcp-client_4.4.3-P1-2_armhf.deb
/root/DEBS/libpam-cap_2.66-4_armhf.deb
/root/DEBS/wireless-tools_30~pre9-14_armhf.deb
/root/DEBS/libiw30_30~pre9-14_armhf.deb
/root/DEBS/libctf0_2.40-2_armhf.deb
/root/DEBS/libnl-route-3-200_3.7.0-0.2+b1_armhf.deb
/root/DEBS/lshw_02.18.85-0.7_armhf.deb
/root/DEBS/libbinutils_2.40-2_armhf.deb
/root/DEBS/libzstd1_1.5.4+dfsg2-5_armhf.deb
/root/DEBS/iproute2_6.1.0-3_armhf.deb

knuxyl, 2024-01-06:
I have done that, but installing basic wifi support is a dependency heirarchy nightmare, and network-manager is just a complete no go. i stopped after collecting about 20 different packages, non stop dependency problems. I think I'll just clone the repo and manually add the nessecary packages for a functioning system.

paoluccigiorgio, 2024-01-07:
Hi, sorry for that....On my mk802 and debian 12.4 image, wifi dependencies (a light nightmare) are resolved with the package list I provided you...I thought they were pretty hardware independent (once got the firmware for your wifi card)....
Best luck.
Giorgio

paoluccigiorgio, 2024-01-07:
Hi, just for sake of completness...In this first bootstrap, you do not need network manager...you just want a device connected to your network..it means have wpa_supplicant working , bring interface up manually with ifconfig and set 8.8.8.8 as dns resolver in your /etc/resolv.conf so that you can do apt-get update and install all the other packages... 
Bests.
Giorgio

knuxyl, 2024-01-07:
I understand that, but if I wanted a do it yourself operating system I would use something like Arch instead, which is what I did after this and the board is running fine. These images aren't plug and play to ssh into, DHCP does not work automatically, so you can't run headless on first boot. Considering the lack of official support for pretty much every single board supported by these images, I would think headless would be the focus for a minimal image, and these images can't do that on initial install. It would be less time consuming to just add the packages needed for proper headless installs and build the image than it would be to find a monitor and keyboard, hook to board, and manually configure all network settings. And I think recommending 8.8.8.8 as a DNS server is terrible advice.

paoluccigiorgio, 2024-01-07:
Well Knuxyl, i understand your point. I wanted just to be helpful..i do not know Rock 3A, but i seemed to me it has an hdmi and usb port...so plugging any recent  tv set and a keyboard, could have solved your problem with little effort...of course 8.8 8.8 is not a good idea, but  not knowing your skills, i suggested  it just to be quick..once having a full functioning system one can configure the dns server he prefers...
Best luck.
Giorgio

knuxyl, 2024-01-07:
There was a long period of time when the board could boot but with no hdmi output. It took a while for linux to support the hdmi on this board, and I'm sure it's not the only one. I don't think you do understand my point. It takes less time and effort to just add the nessecary packages and build a working image than it is to use these images, hook all that stuff up, and manually configure networking.

johang, 2024-01-08:
> These images aren't plug and play to ssh into, DHCP does not work automatically, so you can't run headless on first boot. 

This is wrong and there are [weekly tests](https://github.com/johang/sd-card-images/actions/workflows/prebuilt.yml) to prove that.

Maybe you can post some logs (dmesg, ifconfig) that shows your issue and clarify which boards and images you're using.

knuxyl, 2024-01-08:
> This is wrong and there are [weekly tests](https://github.com/johang/sd-card-images/actions/workflows/prebuilt.yml) to prove that.
>
> Maybe you can post some logs (dmesg, ifconfig) that shows your [...]

I was using debian trixie/testing, and I guarantee you that plugging in ethernet did not automatically get an ip address. And yes, it was hooked directly to a router with DHCP. I switched to Manjaro because it came with packages that worked out of the box, so I can't get and don't have any logs. Maybe it's because I built u-boot myself and installed to SPI, and then debian to emmc? I don't know. But this wasn't the first time this has happened. I built u-boot from the same repo as yours though. Unless you have a Rock 3A and have tested this personally, I'm going to argue these github softwares tests aren't telling you the truth about what is actually happening. I don't know how those tests work, but DHCP is not automatic on this board. Again though, maybe because I have u-boot on SPI. I gave up and I no longer wish to continue looking into it. If I want debian I'll just manually add the needed packages to the build script. Thanks again for this project, I have learned a lot from reading your scripts.

johang, 2024-01-08:
> Maybe it's because I built u-boot myself and installed to SPI, and then debian to emmc? I don't know.

Oh, so you are not even using the published images...

closed 2024-01-08 by johang