johang/sd-card-images #103: RockPro64 doesn't boot with latest uboot + mantic

akhepcat, 2024-01-09:
Using:
* boot-rockpro64.bin.gz 	(2024-01-01)
* ubuntu-mantic-arm64-eemo7y.bin.gz   2024-01-08

written to a 32gb USB drive  (sdcard is currently not recognized, for some reason)

console logs the following:

Found U-Boot script /boot/boot.scr
1287 bytes read in 2 ms (627.9 KiB/s)
## Executing script at 00500000
17364188 bytes read in 414 ms (40 MiB/s)
22404444 bytes read in 532 ms (40.2 MiB/s)
63081 bytes read in 5 ms (12 MiB/s)
kernel_comp_addr_r or kernel_comp_size is not provided!
kernel_comp_addr_r or kernel_comp_size is not provided!
SCRIPT FAILED

mounting the USB drive and decompressing the kernel results in a different error:

Scanning usb 0:2...
Found U-Boot script /boot/boot.scr
1289 bytes read in 2 ms (628.9 KiB/s)
## Executing script at 00500000
55191944 bytes read in 1311 ms (40.1 MiB/s)
22404444 bytes read in 532 ms (40.2 MiB/s)
63081 bytes read in 5 ms (12 MiB/s)
ERROR: RD image overlaps OS image (OS=0x2000000..0x5600000)
ERROR: RD image overlaps OS image (OS=0x2000000..0x5600000)
SCRIPT FAILED:

johang, 2024-01-09:
Can you please show the full log from power on to failure?

akhepcat, 2024-01-09:
It's booting from USB with the latest armbian image just fine, but i'm happy to double-check things.   i've never update the SPI uboot, though.

akhepcat, 2024-01-09:
U-Boot TPL 2023.10-dirty (Jan 01 2024 - 00:11:12)
lpddr4_set_rate: change freq to 400MHz 0, 1
Channel 0: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 800MHz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2023.10-dirty (Jan 01 2024 - 00:11:12 +0000)
Trying to boot from MMC2
## Checking hash(es) for config config-1 ... OK
## Checking hash(es) for Image atf-1 ... sha256+ OK
## Checking hash(es) for Image u-boot ... sha256+ OK
## Checking hash(es) for Image fdt-1 ... sha256+ OK
## Checking hash(es) for Image atf-2 ... sha256+ OK
## Checking hash(es) for Image atf-3 ... sha256+ OK
## Checking hash(es) for Image atf-4 ... sha256+ OK
spl_load_fit_image: Skip load 'atf-5': image size is 0!

U-Boot 2023.10-dirty (Jan 01 2024 - 00:11:12 +0000)

SoC: Rockchip rk3399
Reset cause: POR
Model: Pine64 RockPro64 v2.1
DRAM:  4 GiB (effective 3.9 GiB)
PMIC:  RK808 
Core:  309 devices, 34 uclasses, devicetree: separate
MMC:   mmc@fe310000: 3, mmc@fe320000: 1, mmc@fe330000: 0
Loading Environment from SPIFlash... SF: Detected gd25q128 with page size 256 Bytes, erase size 4 KiB, total 16 MiB
OK
In:    serial@ff1a0000
Out:   serial@ff1a0000
Err:   serial@ff1a0000
Model: Pine64 RockPro64 v2.1
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0 
Card did not respond to voltage select! : -110
Error enabling VMMC supply : -114
Error enabling VMMC supply : -114
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3a0000: USB OHCI 1.0
Bus usb@fe3c0000: USB EHCI 1.00
Bus usb@fe3e0000: USB OHCI 1.0
Bus usb@fe800000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
Bus usb@fe900000: Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3a0000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found
scanning bus usb@fe3e0000 for devices... 1 USB Device(s) found
scanning bus usb@fe800000 for devices... 1 USB Device(s) found
scanning bus usb@fe900000 for devices... cannot reset port 1!?
2 USB Device(s) found
       scanning usb for storage devices... 1 Storage Device(s) found

Device 0: Vendor: General  Rev: 1100 Prod: USB Flash Disk  
            Type: Removable Hard Disk
            Capacity: 123104.0 MB = 120.2 GB (252116992 x 512)
... is now current device
Scanning usb 0:2...
Found U-Boot script /boot/boot.scr
974 bytes read in 2 ms (475.6 KiB/s)
## Executing script at 00500000
17364188 bytes read in 272 ms (60.9 MiB/s)
22404444 bytes read in 350 ms (61 MiB/s)
63081 bytes read in 4 ms (15 MiB/s)
kernel_comp_addr_r or kernel_comp_size is not provided!
kernel_comp_addr_r or kernel_comp_size is not provided!
SCRIPT FAILED

johang, 2024-01-10:
Seems to use the correct U-Boot, but don't understand why kernel_comp_addr_r and kernel_comp_size is not set. Should be by default.

Try going into the U-Boot prompt (at "Hit any key to stop autoboot") and run printenv. It should list all variables set including the kernel_comp_*.