johang/sd-card-images #108: FriendlyElec R2S fails to boot any of Debian images, succeeds with newest Ubuntu

shogsbro, 2024-01-23:
Failing to boot any of the Debian images listed on https://sd-card-images.johang.se/boards/nanopi_r2s.html on FriendlyElec NanoPi R2S V1.0 1912.

I used this [uBoot image, 2024-01-01](https://dl.sd-card-images.johang.se/boots/2024-01-01/boot-nanopi_r2s.bin.gz
) for all tests.

Only one image from that page boots to login prompt; [Ubuntu 24.04 "Noble"](https://dl.sd-card-images.johang.se/debians/2024-01-22/ubuntu-noble-arm64-pouph4.bin.gz). All other images get as far as uBoot starting the kernel, and kernel then fails to negotiate MMC speed and enters initramfs.

---
# Debian Experimental
Excerpts of bootlog from [debian-experimental-arm64-ied9ce.bin.gz](https://dl.sd-card-images.johang.se/debians/2024-01-22/debian-experimental-arm64-ied9ce.bin.gz):
```
[   11.088960] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[   13.219347] dwmmc_rockchip ff500000.mmc: All phases bad!
[
[   13.220308] mmc0: error -5 [   13.240196] mmc_host mmc0: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[   13.348857] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[   13.352625] dwmmc_rockchip ff500000.mmc: All phases bad!
[   13.353133] mmc0: tuning execution failed: -5
[   13.373871] mmc_host mmc0: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[   13.590535] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[   15.722713] dwmmc_rockchip ff500000.mmc: All phases bad!
[   15.723258] mmc0: tuning executi[   15.743543] mmc_host mmc0: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[   15.864093] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[   15.867781] dwmmc_rockchip ff500000.mmc: All phases bad!
[   15.868284] mmc0: tuning execution failed: -5
[   15.889029] mmc_host mmc0: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[   16.115489] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[   18.244840] dwmmc_rockchip ff500000.mmc: All phases bad!
[  itialising SD card
[   18.265457] mmc_host mmc0: Bus speed (slot 0) = 100000Hz (slot req 100000Hz, actual 100000HZ div = 0)
[   18.403587] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[   18.406029] dwmmc_rockchip ff500000.mmc: All phases bad!
[   18.406543] mmc0: tuning execution failed: -5[   18.427356] mmc_host mmc0: Bus speed (slot 0) = 100000Hz (slot req 100000Hz, actual 100000HZ div = 0)
[   18.658208] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[   20.787269] dwmmc_rockchip ff500000.mmc: All phases bad!
```

Noble uBoot printenv
```
=> printenv
arch=arm
baudrate=1500000
board=evb_rk3328
board_name=evb_rk3328
boot_targets=mmc1 mmc0 nvme scsi usb pxe dhcp spi
bootcmtdelay=2
cpu=armv8
cpuid#=00000000000000000000000000000000
et0
fdtcontroladdr=3df21e00
fdtfile=rockchip/rk3328-nanopi-r2s.dnel_comp_size=0x2000000
loadaddr=0x800800
partitions=uuid_disker1};name=loader2,start=8MB,size=4MB,uuid=${uuid_gpt_loader2};na,uuid=${uuid_gpt_boot};name=rootfs,size=-,uuid=B921B045-1DF0-41C000
scriptaddr=0x00500000
serial#=0000000000000000
soc=rk332830000
vendor=rockchip

Environment size: 939/32764 bytes
=> mmc dev 1 0 6
switch to partitions #0, OK
mmc1 is current device
=> mmc info
Device: mmc@ff500000
Manufacturer ID: 3
OEM: 5344
Name: SB32G
Bus Speed: 50000000
Mode: SD High Speed (50MHz)
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 29.7 GiBSize: 512 Bytes
```
---
# Ubuntu Noble
Excerpts of bootlog from [ubuntu-noble-arm64-pouph4.bin.gz](https://dl.sd-card-images.johang.se/debians/2024-01-22/ubuntu-noble-arm64-pouph4.bin.gz):
```
[    4.870621] dwmmc_rockchip ff500000.mmc: IDMAC supports 32-bit address mode.
[    4.871612] dwmmc_rockchip ff500000.mmc: Using internal DMA controller.
[    4.872291] dwmmc_rockchip ff500is 270a
[    4.873004] dwmmc_rockchip ff500000.mmc: DW MMC contdth,256 deep fifo
Begin: Mounting root file system ... [    4.893719] mmc_host mmc0: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    5.000949] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[    5.165995] dwmmc_rockchip ff500000.mmc: Successfully tuned phase to 10
[    5.166724] mmc0: new ultra high speed SDR104 SDHC card at address aaaa
[    5.172465] mmcblk0: mmc0:aaaa SB32G 29.7 GiB
[    5.182046]  mmcblk0: p1 p2
Warning: fsck not present, so skipping root file system
[   35.211839] EXT4-fs (mmcblk0p2): recovery complete
[   35.213760] EXT4-fs (mmcblk0p2): mounted filesystem d6ac97a1-75f2-471a-9659-faba7175900c r/w with ordeata mode. Quota mode: none.
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... done.
```

Noble uBoot printenv
```
=> printenv
arch=arm
baudrate=1500000
board=evb_rk3328
board_name=evb_rk3328
boot_targets=mmc1 mmc0 nvme scsi usb pxe dhcp spi
bootcm=armv8
cpuid#=00000000000000000000000000000000
eth1addr=86:e0:addr=3df21e00
fdtfile=rockchip/rk3328-nanopi-r2s.dtb
kernel_ad=0x2000000
loadaddr=0x800800
partitions=uuid_disk=${uuid_gpt_der2,start=8MB,size=4MB,uuid=${uuid_gpt_loader2};name=trust,size=4M,uuid=${uuid_gpt_atf};name=boot,size=112M,bootable,uuid=${uuid_gpt_boot};name=rootfs,size=-,uuid=B921B045-1DF0-41C3-AF44-4C6F2dr=0x00500000
serial#=0000000000000000
soc=rk3328
stderr=seri=rockchip

Environment size: 939/32764 bytes
=> bdinfo
boot_params = 0x0000000000000000
DRAM bank   = 0x0000000000000000
-> start    = 0x0000000000200000
-> size     = 0x00000000flashstart  = 0x0000000000000000
flashsize   = 0x00000000000000relocaddr   = 0x000000003ff34000
reloc off   = 0x000000003fd340= 86:e0:c0:ea:fa:a9
IP addr     = <NULL>
fdt_blob    = 0x0000000000000009fc0
lmb_dump_all:
 memory.cnt = 0x1 / max = 0x10
 memory[0]	[0x200000-0x3fffffff], 0x3fe00000 bytes flags: 0
 resnt = 0x2 / max = 0x10
 reserved[0]	[0x3cefd000-0x3fffffff], 0x010 bytes flags: 0
devicetree  = separate
serial addr = 0x00000000000000002
 offset     = 0x0000000000000000
 clock      = 0x0000003fff0000
irq_sp      = 0x000000003df21df0
sp start    =
```

[ubuntu-noble-printenv.log](https://github.com/johang/sd-card-images/files/14016943/ubuntu-noble-printenv.log)
[ubuntu-noble.log](https://github.com/johang/sd-card-images/files/14016945/ubuntu-noble.log)
[debian-experimental.log](https://github.com/johang/sd-card-images/files/14016946/debian-experimental.log)
[debian-experimental-printenv.log](https://github.com/johang/sd-card-images/files/14016947/debian-experimental-printenv.log)

---
# Footnote
I previously downloaded a set of working Bookworm files, that I have running on another similar vintage NanoPi R2S, but they are unfortunately now lost. 
```
root@nanopi3.hh:~# inxi -S
System:
  Host: nanopi3.hh Kernel: 6.1.0-17-arm64 arch: aarch64 bits: 64 Console: pty pts/0 Distro: Debian
    GNU/Linux 12 (bookworm)
root@nanopi3.hh:~# inxi -D
Drives:
  Local Storage: total: 28.98 GiB used: 4.03 GiB (13.9%)
  ID-1: /dev/mmcblk0 vendor: Toshiba model: SA32G size: 28.98 GiB
```

jwetzell, 2024-08-10:
I'm unable to get any image to boot for the Nano Pi R2S. I cannot test with that 2024-01-01 boot image as it isn't available anymore I only have access to the 2024-08-01 boot image that is currently up.

fernvenue, 2024-12-20:
Same here, I found that even if it starts successfully, the SD card speed on Debian is high speed instead of ultra high speed, and there are system interruptions and unexpected restarts, often making it impossible to boot.

shogsbro, 2025-11-30:
I think this can be closed now. I've just installed these on 3 x Nanopi R2S:
- curl -O https://dl.sd-card-images.johang.se/boots/2025-11-01/boot-nanopi_r2s.bin.gz
- curl -O https://dl.sd-card-images.johang.se/debians/2025-11-24/debian-trixie-arm64-pe9ami.bin.gz