johang/sd-card-images #119: OrangePi One Plus No ethernet found

SecondCap, 2024-02-20:
PHY Type： RTL8211

```
U-Boot 2024.01johang-dirty (Feb 01 2024 - 02:19:38 +0000) Allwinner Technology

CPU:   Allwinner H6 (SUN50I)
Model: OrangePi One Plus
DRAM:  1 GiB
Core:  55 devices, 17 uclasses, devicetree: separate
WDT:   Not starting watchdog@7020400
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to use mmc 0:2...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
starting USB...
Bus usb@5101000: USB EHCI 1.00
Bus usb@5101400: USB OHCI 1.0
Bus usb@5311000: USB EHCI 1.00
Bus usb@5311400: USB OHCI 1.0
scanning bus usb@5101000 for devices... 1 USB Device(s) found
scanning bus usb@5101400 for devices... 1 USB Device(s) found
scanning bus usb@5311000 for devices... 1 USB Device(s) found
scanning bus usb@5311400 for devices... 1 USB Device(s) found
scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
```

johang, 2024-02-20:
This is just U-Boot built without ethernet support.

SecondCap, 2024-02-21:
Sorry, at first I thought it was a uboot issue. 
On debian images(bulleye,bookworm,experimental), the kernel prints:

```context
[   12.471925] dwmac-sun8i 5020000.ethernet: IRQ eth_lpi not found
[   12.491215] dwmac-sun8i 5020000.ethernet: PTP uses main clock
[   12.497871] dwmac-sun8i 5020000.ethernet: Current syscon value is not the default 58000 (expect 50000)
[   12.613838] dwmac-sun8i 5020000.ethernet: No HW DMA feature register supported
[   12.621142] dwmac-sun8i 5020000.ethernet: RX Checksum Offload Engine supported
[   12.628491] dwmac-sun8i 5020000.ethernet: COE Type 2
[   12.633529] dwmac-sun8i 5020000.ethernet: TX Checksum insertion supported
[   12.633543] dwmac-sun8i 5020000.ethernet: Normal descriptors
[   12.633549] dwmac-sun8i 5020000.ethernet: Chain mode enabled
[   12.634916] mdio_bus stmmac-0: MDIO device at address 1 is missing.
[   12.704624] dwmac-sun8i 5020000.ethernet end0: renamed from eth0
[   12.827187] dwmac-sun8i 5020000.ethernet end0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   12.836347] dwmac-sun8i 5020000.ethernet end0: __stmmac_open: Cannot attach to PHY (error: -19)
```

SecondCap, 2024-02-23:
> This is just U-Boot built without ethernet support.

Sorry, at first I thought it was a uboot issue. 
On debian images(bulleye,bookworm,experimental), the kernel prints:

```context
[   12.471925] dwmac-sun8i 5020000.ethernet: IRQ eth_lpi not found
[   12.491215] dwmac-sun8i 5020000.ethernet: PTP uses main clock
[   12.497871] dwmac-sun8i 5020000.ethernet: Current syscon value is not the default 58000 (expect 50000)
[   12.613838] dwmac-sun8i 5020000.ethernet: No HW DMA feature register supported
[   12.621142] dwmac-sun8i 5020000.ethernet: RX Checksum Offload Engine supported
[   12.628491] dwmac-sun8i 5020000.ethernet: COE Type 2
[   12.633529] dwmac-sun8i 5020000.ethernet: TX Checksum insertion supported
[   12.633543] dwmac-sun8i 5020000.ethernet: Normal descriptors
[   12.633549] dwmac-sun8i 5020000.ethernet: Chain mode enabled
[   12.634916] mdio_bus stmmac-0: MDIO device at address 1 is missing.
[   12.704624] dwmac-sun8i 5020000.ethernet end0: renamed from eth0
[   12.827187] dwmac-sun8i 5020000.ethernet end0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   12.836347] dwmac-sun8i 5020000.ethernet end0: __stmmac_open: Cannot attach to PHY (error: -19)
```

johang, 2024-02-23:
Seems like it can't find the PHY at the address 1.

closed 2024-03-12 by SecondCap