johang/sd-card-images #122: Missing HDMI output on BananaPi M5 with last bookworm image

berni51, 2024-03-01:
Hi there,
on my third BananaPi M5 I need a stable Linux image. So I downloaded the last johang bookworm image, wrote it to a sd card and started.
Unfortunately my Monitor stayed dark, nothing, absoluety nothing to see. However the monitor must get a signal, because the blue monitor LED was permenetly on. But there was nothing to see on the screen.
I tried to connect to the new BananaPi M% via ssh, and that works perfect. I was able to log in as root and could do all my settings (keyboard, installing apps)
Possibly there is only a small step I missed to do, but have no idea what this step may be,
Any hints for me?

Kind regards
Berni

johang, 2024-03-01:
These images are very minimal on purpose. You shouldn't expect advanced features to work out of the box.

berni51, 2024-03-01:
Minimal is perfect for me to build a camera server with the Pi. But some characters on the screen are a must for me. :=)

Berni

johang, 2024-03-07:
Try the the other images. They have newer kernels.