johang/sd-card-images #125: Devuan - Debian without systemd images?

anton-, 2024-03-15:
Hi, 
Could be possible to add Devuan distro (https://devuan.org) repositories to generate Devuan images? Many of these devices can be used without desktop environment and in these cases to have Systemd installed have no much use.

Thanks for your great work.

watchful-0wl, 2024-05-17:
I added Devuan to a fork of johang/sd-card-images at [https://github.com/watchful-0wl/sd-card-images](https://github.com/watchful-0wl/sd-card-images).  There are no pre-built Devuan images there, it is for DIY Devuan builds.

[README_devuan.md](https://github.com/watchful-0wl/sd-card-images/blob/master/README_devuan.md).  With it, I have built and booted Devuan Daedalus on a Pine64 RockPro64.

anton-, 2024-05-21:
Hi,
Thanks for your work, I will try to generate it for my old arm stick. I
really appreciate.
Antonio