johang/sd-card-images #130: NanoPC T6 not booting

J-Twin, 2024-04-27:
Hello,

I have installed the noble and trixie images to a known good SD card but neither image will boot on my T6. There is nothing displayed at the screen which is using HDMI cable. Same SD card is booting Armbian running kernel 6.1. I have successfully installed Armbian at the attached NVME disk.

I cannot get any logs as it does not actually boot.

johang, 2024-04-27:
Connect to the UART console and show the complete log output during boot.

Starbane, 2024-05-12:
I experience the same thing, it will not boot. I try to mount the boot image and get an error stating that there is some thing wrong with /dev/loop

J-Twin, 2024-05-13:
> Connect to the UART console and show the complete log output during boot.

Things are not looking good at all! System gets in a boot loop shortly after loading the kernel:

DDR Version V1.08 20220617
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
Manufacturer ID:0x1 Samsung
CH0 RX Vref:30.7%, TX Vref:21.8%,22.8%
CH1 RX Vref:30.7%, TX Vref:23.8%,22.8%
CH2 RX Vref:31.7%, TX Vref:21.8%,21.8%
CH3 RX Vref:29.7%, TX Vref:23.8%,22.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-armbian (Mar 06 2024 - 00:28:07)
Trying to boot from MMC2
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: ***        Using Backup GPT ***
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(b40ce543bb...) + OK
## Checking u-boot 0x00a00000 ... sha256(c51a050e8c...) + OK
## Checking fdt-1 0x00ace978 ... sha256(feae82f1bf...) + OK
## Checking atf-2 0x000f0000 ... sha256(f202de33d9...) + OK
## Checking atf-3 0xff100000 ... sha256(af08976afd...) + OK
Jumping to U-Boot(0x00a00000) via ARM Trusted Firmware(0x00040000)
Total: 851.327 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-616-gdc1125f48:finley.xiao
NOTICE:  BL31: Built : 16:41:17, Jul 13 2023
INFO:    spec: 0x1
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
ERROR:   dfs get fsp_params[0] error, 0xfead0001 != 0xfead0003
ERROR:   dfs get fsp_params[1] error, 0xa4c != 0xfead0003
ERROR:   dfs get fsp_params[2] error, 0xa44 != 0xfead0003
ERROR:   dfs get fsp_params[3] error, 0x4000001 != 0xfead0003
ERROR:   loader&trust unmatch!!! Please update loader if need enable dmc
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2024.01johang-dirty (Apr 01 2024 - 02:14:17 +0000)

Model: FriendlyElec NanoPC-T6
DRAM:  16 GiB (effective 15.7 GiB)
Core:  332 devices, 28 uclasses, devicetree: separate
MMC:   mmc@fe2c0000: 1, mmc@fe2e0000: 0
Loading Environment from nowhere... OK
In:    serial@feb50000
Out:   serial@feb50000
Err:   serial@feb50000
Model: FriendlyElec NanoPC-T6
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
** Booting bootflow 'mmc@fe2c0000.bootdev.part_2' with script
18199471 bytes read in 1524 ms (11.4 MiB/s)
24245737 bytes read in 2030 ms (11.4 MiB/s)
73885 bytes read in 21 ms (3.4 MiB/s)
   Uncompressing Kernel Image
Moving Image from 0x2080000 to 0x2200000, end=5a70000
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
Moving Image from 0x2080000 to 0x2200000, end=5a70000
## Flattened Device Tree blob at edbe56f0
   Booting using the fdt blob at 0xedbe56f0
Working FDT set to edbe56f0
   Loading Ramdisk to eb4ba000, end ecbd95e9 ... OK
   Loading Device Tree to 00000000eb49c000, end 00000000eb4b9687 ... OK
Working FDT set to eb49c000

Starting kernel ...

"Synchronous Abort" handler, esr 0x96000010, far 0x1f0
elr: ffffffff153100a0 lr : 0000000000a027f0 (reloc)
elr: 00000000048220a0 lr : 00000000eff147f0
x0 : 00000000eb49c000 x1 : 0000000000000000
x2 : 0000000000000000 x3 : 0000000000000000
x4 : 0000000002200000 x5 : 0000000000000001
x6 : 0000000000000008 x7 : 0000000000000000
x8 : 00000000000000a0 x9 : 0000000000000004
x10: 00000000c3000123 x11: 0000000000000003
x12: 0000000000000004 x13: 00000000eb49c000
x14: 00000000ffffffff x15: 00000000eff13784
x16: 00000000eff60590 x17: 0000000000000000
x18: 00000000edbffd90 x19: 00000000effe0a18
x20: 0000000000000000 x21: 0000000000000000
x22: 0000000000000003 x23: 00000000edf36008
x24: 00000000effc9800 x25: 0000000000000000
x26: 00000000eff14818 x27: 0000000000000000
x28: 0000000000000400 x29: 00000000edbe4c40

Code: 72676574 36383161 636d655f 6265645f (6d5f6775) 
Resetting CPU ...

resetting ...

This goes over and over forever!

There is a mismatch compared to working Armbian image using vendor kernel 6.1.43:
ERROR:   dfs get fsp_params[0] error, 0xfead0001 != 0xfead0003
ERROR:   dfs get fsp_params[1] error, 0xa4c != 0xfead0003
ERROR:   dfs get fsp_params[2] error, 0xa44 != 0xfead0003
ERROR:   dfs get fsp_params[3] error, 0x4000001 != 0xfead0003
ERROR:   loader&trust unmatch!!! Please update loader if need enable dmc

Armbian:
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz

Image used is as4ahb

johang, 2024-05-15:
```
18199471 bytes read in 1524 ms (11.4 MiB/s)
24245737 bytes read in 2030 ms (11.4 MiB/s)
73885 bytes read in 21 ms (3.4 MiB/s)
Uncompressing Kernel Image
Moving Image from 0x2080000 to 0x2200000, end=5a70000
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
```

This is the problem. DTB is corrupt for some reason.

J-Twin, 2024-05-15:
Is there a way to dig into this?

SD card used to write the image file is in good condition. The same card was used to boot another noble image with success:
https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.1.0/ubuntu-24.04-preinstalled-desktop-arm64-nanopc-t6.img.xz

That image has the "dfs get fsp_params" error too but boots normally

johang, 2024-05-15:
I would go into the U-Boot console at `Hit any key to stop autoboot: 0`, run the boot script manually and then inspect the contents of at the DTB address.

J-Twin, 2024-06-08:
Hello, I have updated bootloader to v1.16 from v1.08 but problem remains.

This bootloader along with friendlyelec U-Boot SPL are able to boot properly all Armbian SD images.