johang/sd-card-images #138: Cubieboard4 freeze on boot

som1712, 2024-07-15:
Hello
Cubieboard 4 does not start, as seen in the log below:

U-Boot SPL 2024.04johang-dirty (Jul 01 2024 - 02:33:09 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1

U-Boot 2024.04johang-dirty (Jul 01 2024 - 02:33:09 +0000) Allwinner Technology

CPU: Allwinner A80 (SUN9I)
Model: Cubietech Cubieboard4
DRAM: 2 GiB
Core: 59 devices, 17 uclasses, devicetree: separate
WDT: Not starting watchdog@6000ca0
WDT: Not starting watchdog@8001000
MMC: sunxi_set_reset: (RST#0) unhandled
sunxi_set_reset: (RST#2) unhandled
sunxi_set_reset: (RST#1) unhandled
mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 0:2...
In: serial@7000000
Out: serial@7000000
Err: serial@7000000
Net: No ethernet found.
Hit any key to stop autoboot: 0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:2...
Found U-Boot script /boot/boot.scr
968 bytes read in 2 ms (472.7 KiB/s)
## Executing script at 23100000
4997632 bytes read in 216 ms (22.1 MiB/s)
10679119 bytes read in 461 ms (22.1 MiB/s)
24534 bytes read in 41 ms (584 KiB/s)
Kernel image @ 0x22000000 [ 0x000000 - 0x4c4200 ]
## Flattened Device Tree blob at 23000000
 Booting using the fdt blob at 0x23000000
Working FDT set to 23000000
 Loading Ramdisk to 295d0000, end 29fff34f...OK
 Loading Device Tree to 295c7000, end 295cffd5 ... OK
Working FDT set to 295c7000

Starting kernel...

The last build I tested and works has U-Boot 2022.04.
Is it possible to make the image available again for download?

johang, 2024-07-17:
You can build your own image with the version you want.

Bart4711, 2025-05-18:
I've just tried it again with the latest images and my Cubieboard4 is booting up and I'm able to log in via ssh.
Boot image from 2025-05-01
Debian stable/12.10 ("bookworm") image from 2025-05-12
So this issue seems to be fixed. Thank you!