johang/sd-card-images #144: Several different issues with Amlogic P212

LupusEtFlos, 2024-09-03:
Hello. I've followed the instructions for Debian Bookworm, using a 16GB SD card, but the dd process takes several minutes, prints out an error about input/output, but eventually finishes. I end up with a 29MB FAT16 partition, a 3.5GB bootable partition with an unrecognisable filesystem, and the rest is unallocated.

I then tried to insert the card on my device, and it boots straight to Android, the card shows up fine under the storage menu. Am I doing something wrong, could I attempt maybe Balena Etcher instead of dd to flash the image?

Worth mentioning, this board belonged to my old ISP, they used it as a streaming device I managed to escape from and reach the Android OS, I wonder if I need to flash its stock ROM first?

Any help or guidance is truly appreciated.

Here's screenshots.
[image]
[image]

johang, 2024-09-05:
>  but the dd process [...] prints out an error about input/output

This might be the problem. Bad SD card?

LupusEtFlos, 2024-09-09:
> This might be the problem. Bad SD card?

Hmm, it works fine otherwise. I don't have another card big enough to test with. To be honest, every SD card I try to format with dd, for other projects I mean, give me issues of all sorts. I'm starting to wonder if the interface I'm using is the issue, or the adaptor. But... I don't have any other devices to do this process with.

Thanks for answering, by the way. I hope to be able to run this on my box someday.