johang/sd-card-images #150: Orange Pi Zero 3 does not boot

knuxyl, 2024-10-15:
Board hangs with no output on UART and does not receive IP from DHCP server so I'm guessing kernel is not even loading. Waited 15 minutes and it was still saying "Starting kernel ..."

> U-Boot SPL 2024.07johang-dirty (Oct 01 2024 - 02:23:24 +0000)                   
> DRAM: 2048 MiB                                                                  
> Trying to boot from MMC1                                                        
> NOTICE:  BL31: v2.9(debug):v2.9.0                                               
> NOTICE:  BL31: Built : 02:23:17, Oct  1 2024                                    
> NOTICE:  BL31: Detected Allwinner H616 SoC (1823)                               
> NOTICE:  BL31: Found U-Boot DTB at 0x4a0b3d18, model: OrangePi Zero3            
> INFO:    ARM GICv2 driver initialized                                           
> INFO:    Configuring SPC Controller                                             
> INFO:    PMIC: Probing AXP305 on RSB                                            
> ERROR:   RSB: set run-time address: 0x10003                                     
> INFO:    Could not init RSB: -65539                                             
> INFO:    BL31: Platform setup done                                              
> INFO:    BL31: Initializing runtime services                                    
> INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied                
> INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied               
> INFO:    PSCI: Suspend is unavailable                                           
> INFO:    BL31: Preparing for EL3 exit to normal world                           
> INFO:    Entry point address = 0x4a000000                                       
> INFO:    SPSR = 0x3c9                                                           
> INFO:    Changed devicetree.                                                    
>                                                                                 
>                                                                                 
> U-Boot 2024.07johang-dirty (Oct 01 2024 - 02:23:24 +0000) Allwinner Technology  
>                                                                                 
> CPU:   Allwinner H616 (SUN50I)                                                  
> Model: OrangePi Zero3                                                           
> DRAM:  2 GiB                                                                    
> Core:  58 devices, 25 uclasses, devicetree: separate                            
> WDT:   Not starting watchdog@30090a0                                            
> MMC:   mmc@4020000: 0                                                           
> Loading Environment from FAT... Unable to use mmc 0:2...                        
> In:    serial@5000000                                                           
> Out:   serial@5000000                                                           
> Err:   serial@5000000                                                           
> Allwinner mUSB OTG (Peripheral)                                                 
> Net:   eth0: ethernet@5020000using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in  
> MAC de:ad:be:ef:00:01                                                           
> HOST MAC de:ad:be:ef:00:00                                                      
> RNDIS ready                                                                     
> , eth1: usb_ether                                                               
> starting USB...                                                                 
> Bus usb@5200000: USB EHCI 1.00                                                  
> Bus usb@5200400: USB OHCI 1.0                                                   
> scanning bus usb@5200000 for devices... 1 USB Device(s) found                   
> scanning bus usb@5200400 for devices... 1 USB Device(s) found                   
>        scanning usb for storage devices... 0 Storage Device(s) found            
> Hit any key to stop autoboot:  0                                                
> switch to partitions #0, OK                                                     
> mmc0 is current device                                                          
> Scanning mmc 0:2...                                                             
> Found U-Boot script /boot/boot.scr                                              
> 956 bytes read in 2 ms (466.8 KiB/s)                                            
> ## Executing script at 4fc00000                                                 
> 32968640 bytes read in 1315 ms (23.9 MiB/s)                                     
> 15004753 bytes read in 623 ms (23 MiB/s)                                        
> 19655 bytes read in 6 ms (3.1 MiB/s)                                            
> Moving Image from 0x40080000 to 0x40200000, end=42240000                        
> ## Flattened Device Tree blob at 4fa00000                                       
>    Booting using the fdt blob at 0x4fa00000                                     
> Working FDT set to 4fa00000                                                     
>    Loading Ramdisk to 491b0000, end 49fff451 ... OK                             
>    Loading Device Tree to 00000000491a8000, end 00000000491afcc6 ... OK         
> Working FDT set to 491a8000                                                     
>                                                                                 
> Starting kernel ...

It looks like this is using the older method of booting with boot.scr. I don't know if bootmeth extlinux would work with this board though. My old boot.scr for booting arch linux on this had these contents and worked fine

> ext2load mmc 0 0x46000000 boot/Image
> ext2load mmc 0 0x49000000 sun50i-h618-orangepi-zero3.dtb
> setenv bootargs console=ttyS0,115200 root=/dev/mmcblk0p1 rw nodiratime noatime rootwait panic=10 ${extra}
> booti 0x46000000 - 0x49000000

liaoya, 2024-12-23:
I have the same issue with the board (my board has 1.5G Memory). The following is the UART output on my board (Then the output hangs). At least two problems

1. The CPU showed is `Allwinner H616`
2. The memory is `DRAM:  2 GiB`

```text
U-Boot 2024.10johang-dirty (Dec 01 2024 - 02:28:38 +0000) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero3
DRAM:  2 GiB (effective 13 EiB)

U-Boot SPL 2024.10johang-dirty (Dec 01 2024 - 02:28:38 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.9(debug):v2.9.0
NOTICE:  BL31: Built : 02:28:22, Dec  1 2024
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0b4518, model: OrangePi Zero3
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.
```

sukibaby, 2025-04-15:
If this still occurs, it may be an issue specific to 1.5GB boards. I have a 2GB board and recent images are working.

-------------

journalctl -b
```
Feb 21 21:18:32 ubuntu kernel: Memory: 1939396K/2097152K available (23104K kernel code, 5190K rwdat>
Feb 21 21:18:32 ubuntu kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Feb 21 21:18:32 ubuntu kernel: ftrace: allocating 62662 entries in 245 pages
Feb 21 21:18:32 ubuntu kernel: ftrace: allocated 245 pages with 6 groups
lines 1-48
Feb 21 21:18:32 ubuntu kernel: Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Feb 21 21:18:32 ubuntu kernel: Linux version 6.8.0-57-generic (buildd@bos03-arm64-086) (aarch64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubunt>
Feb 21 21:18:32 ubuntu kernel: KASLR disabled due to lack of seed
Feb 21 21:18:32 ubuntu kernel: Machine model: OrangePi Zero3
```

uname -a
```
Linux ubuntu 6.8.0-57-generic #59-Ubuntu SMP PREEMPT_DYNAMIC Sat Mar 15 17:00:30 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
```

liaoya, 2025-04-18:
It‘s  an issue specific to 1.5GB board. OpenWrt `24.10.0` also has such kind of issue. <https://github.com/openwrt/openwrt/issues/18087>. The difference is OpenWrt use Uboot `2024.01`. But Armbian is OK and it use Uboot `2024.01` also. you can find this in <https://github.com/openwrt/openwrt/issues/18087>. It's definitely Uboot issue. Hope upstream can fix this issue.

fhersete, 2025-04-18:
> of issue. openwrt/openwrt#18087
> also. you can find this in openwrt/openwrt#18087

Check if you have added the dtb for the uboot boot of your OPI Zero3 version.

liaoya, 2025-05-28:
@fhersete 
I'm sorry I can not understand your instruction.

liaoya, 2025-05-28:
I tried lastest sd-card-image build and upstream uboot `2025.04` still does not recognize the 1.5G memory.

fhersete, 2025-05-28:
https://github.com/MichaIng/DietPi/issues/6938

WangBen137, 2025-08-20:
那各位有什么解决方案吗