johang/sd-card-images #154: OrangePi Zero2+ (H5) keyboard not working

Drout, 2024-11-23:
I've tried with several images, the system boots up real quick, but at the login prompt I can't type anything.
Keyboard works for sure (with armbian and orangepi images).

liaoya, 2024-12-23:
@Drout Can you try the following. That's work on my zero 2 (https://sd-card-images.johang.se/boards/orange_pi_zero2.html). I use UART and micro HDMI does not work.

```sh
wget https://dl.sd-card-images.johang.se/boots/2024-12-01/boot-orange_pi_zero2.bin.gz
wget https://dl.sd-card-images.johang.se/debians/2024-12-16/ubuntu-noble-arm64-zoze7f.bin.gz
zcat boot-orange_pi_zero2.bin.gz ubuntu-noble-arm64-zoze7f.bin.gz > sd-card.img
dd if=sd-card.img of=/dev/sdXXX # ...where /dev/sdXXX is your SD card
```

liaoya, 2024-12-23:
The problem I met is that the OS will think the board has 2G memory sometime

[image]