johang/sd-card-images #155: Reboot fails on Ubuntu noble and oracular

LupeChristoph, 2024-11-28:
I'm using a Nanopi R4S which alas, has no serial console, so I can't really debug this.

Status of the Ubuntu images:
jammy reboots OK
noble hangs with the power LED on, no network
oracular same as noble
plucky same

Incidentally, Debian bookworm is OK, trixie starts both ethernets but is not accessible (nmap says "All 1000 scanned ports on 172.17.0.129 are in ignored states." on both ethernet interfaces). I gave up after that. But I will open a ticket for this.

OpenWRT from their download is OK with 23.03.3 and .4.

LupeChristoph, 2024-11-30:
The problem with trixie was due to a bad microSD card.

Reboot failures with Debian, using a good microSD card:
bookworm: works OK
trixie: fails, SYS LED stays off
sid: same as trixie
rc-buggy: same as trixie

The strange thing is that trixie, sid and rc-buggy are not accessible with SSH after the first boot of an image. After the second or third boot, SSH is available. As with reboot, bookworm is OK.

LupeChristoph, 2024-11-30:
I've acquired a USB-to-serial adapter and dismantled the R4S. I can now see the console output, and I found the problem.

```
[  OK  ] Reached target reboot.target - System Reboot.
[  117.632275] reboot: Restarting system

U-Boot TPL 2024.10johang-dirty (Nov 01 2024 - 02:10:40)
lpddr4_set_rate: change freq to 400MHz 0, 1
Channel 0: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 800MHz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2024.10johang-dirty (Nov 01 2024 - 02:10:40 +0000)
Trying to boot from MMC2
mmc_load_image_raw_sector: mmc block read error
Trying to boot from MMC2
mmc_load_image_raw_sector: mmc block read error
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
```

Nothing after that. I doubt the Linux kernel can do anything about that :-( But i wonder why jammy has no problem with that. I;ll run more tests to see if this happens randomly.

LupeChristoph, 2024-12-01:
[Launchpad ticket #2090788](https://bugs.launchpad.net/ubuntu/+source/wireless-tools/+bug/2090788) _On a NanoPi R4S, oracular fails the first reboot_

LupeChristoph, 2024-12-01:
I just checked if a normal power down would also fix the reboot problem, and indeed it does. After a power down , the reboot works.

LupeChristoph, 2024-12-01:
I just checked my old copy of oracular and that is permamently stuck with the reboot problem. Strange. I'll keep this card for further investigations and bring a new copy of oracular up. Fortunately I wasn't *that* far into setting up...

johang, 2024-12-01:
Can you please post a full log of a reboot when it succeeds?

LupeChristoph, 2024-12-01:
Here it is: [wart-reboot.log](https://github.com/user-attachments/files/17970320/wart-reboot.log)

johang, 2024-12-02:
Yeah, it's probably U-Boot not resetting something.

LupeChristoph, 2024-12-02:
I wondered why my first microSD card continues to have problems rebooting. And I remembered that I had expanded the root filesystem to the and of the card with this command (from your FAQ):
`parted -a opt /dev/sdXXX "resizepart 2 100%"`
So I did that again, but again neglected to invoke partprobe. And the R4S did not reboot.

I restored the boot block from the `boot-nanopi_r4s.bin.gz` file and booted the R4S again. This time I invoked partprobe after parted. No change. And the reboot never works.

To repeat - just changing the end address of partition 2 makes the reboot problem persist. I did not expand the filesystem, just the partition.

To debug better I connected the USB serial adapter. Log to follow.

LupeChristoph, 2024-12-03:
This is really hard to reproduce. I'm giving up for now. More when I have an idea what might trigger the hang.

LupeChristoph, 2024-12-05:
Looks like I did too much work on this. I'm very susceptible to stress, and this was too much. Sorry, I'll have to take some time off to recover.

LupeChristoph, 2025-01-12:
Sorry that I took so long to return. First thing I'm testing the current ISOs for jammy and oracular to see if I can add a data partition and still reboot. I'll test jammy first, as that did not exhibit the reboot problem.

It is OK. No reboot problem.

Now it gets interesting. Ubuntu oracular does not have the reboot problem on the first reboot after installation anymore. Creating a partition and a filesystem on it does not change that.

Looks like you fixed the reboot problem while I was recharging my internal battery. Thank you very much for that.

closed 2025-01-12 by LupeChristoph