johang/sd-card-images #174: odroid M1 build does not work

The-Hiker, 2025-02-12:
these instructions:

wget https://dl.sd-card-images.johang.se/boots/2025-02-01/boot-odroid_m1.bin.gz
wget https://dl.sd-card-images.johang.se/debians/2025-02-10/ubuntu-noble-arm64-heich4.bin.gz
zcat boot-odroid_m1.bin.gz ubuntu-noble-arm64-heich4.bin.gz > sd-card.img
dd if=sd-card.img of=/dev/sdXXX # ...where /dev/sdXXX is your SD card

Do not produce a bootable disk for an Odroid M1

The-Hiker, 2025-02-14:
this is the output to the console before it locks-up:

DDR Version V1.09 20210630
In
ddrconfig:7                                                                                                                
LPDDR4X, 324MHz                                                                                                            
BW=32 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=8192MB                                                          
tdqss: cs0 dqs0: 48ps, dqs1: -120ps, dqs2: -96ps, dqs3: -144ps,                                                            
tdqss: cs1 dqs0: -24ps, dqs1: -120ps, dqs2: -72ps, dqs3: -168ps,                                                           
                                                                                                                           
change to: 324MHz                                                                                                          
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:41%                                                                                                 
dram drv:40,odt:0                                                                                                          
clk skew:0x62                                                                                                              
                                                                                                                           
change to: 528MHz                                                                                                          
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:41%                                                                                                 
dram drv:40,odt:0                                                                                                          
clk skew:0x58                                                                                                              
                                                                                                                           
change to: 780MHz                                                                                                          
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:41%                                                                                                 
dram drv:40,odt:0                                                                                                          
clk skew:0x58                                                                                                              
                                                                                                                           
change to: 1560MHz(final freq)                                                                                             
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:22%                                                                                                 
dram drv:40,odt:80                                                                                                         
vref_ca:00000071                                                                                                           
clk skew:0x26                                                                                                              
cs 0:                                                                                                                      
the read training result:                                                                                                  
DQS0:0x2e, DQS1:0x30, DQS2:0x32, DQS3:0x29,                                                                                
min  : 0xe  0xf 0x12  0xf  0x2  0x5  0x9  0x3 , 0xa  0x8  0x2  0x1  0xf  0xb  0xd  0x9 ,                                   
      0x11 0x12  0xc  0xb  0x5  0x3  0x3  0x5 , 0xa  0x8  0x8  0x1  0xf  0xd  0xb  0xd ,                                   
mid  :0x27 0x29 0x2b 0x29 0x1c 0x20 0x22 0x1e ,0x23 0x22 0x1b 0x1b 0x27 0x23 0x24 0x21 ,                                   
      0x2b 0x2c 0x26 0x24 0x1e 0x1b 0x1d 0x1f ,0x23 0x1f 0x1f 0x1a 0x27 0x27 0x24 0x26 ,                                   
max  :0x40 0x43 0x44 0x44 0x37 0x3b 0x3c 0x39 ,0x3d 0x3d 0x34 0x36 0x3f 0x3b 0x3c 0x3a ,                                   
      0x45 0x47 0x40 0x3e 0x38 0x34 0x37 0x39 ,0x3d 0x37 0x37 0x34 0x40 0x41 0x3d 0x3f ,                                   
range:0x32 0x34 0x32 0x35 0x35 0x36 0x33 0x36 ,0x33 0x35 0x32 0x35 0x30 0x30 0x2f 0x31 ,                                   
      0x34 0x35 0x34 0x33 0x33 0x31 0x34 0x34 ,0x33 0x2f 0x2f 0x33 0x31 0x34 0x32 0x32 ,                                   
the write training result:                                                                                                 
DQS0:0x2f, DQS1:0xf, DQS2:0x13, DQS3:0xa,                                                                                  
[... cut ...]
[    1.338452] rkflash_dev_init device register as mtd dev, ret= 0                                                         
[    1.345946] rk_gmac-dwmac fe2a0000.ethernet: no regulator found                                                         
[    1.352391] rk_gmac-dwmac fe2a0000.ethernet: rk_get_eth_addr: mac address: 00:1e:06:51:0d:30                            
                                                                                                                           
/ #

johang, 2025-02-15:
Looks like it's booting something else. Not these images.

The-Hiker, 2025-02-16:
Ah yes, when the noble build didn't work i tried the oracular build:

wget https://dl.sd-card-images.johang.se/boots/2025-02-01/boot-odroid_m1.bin.gz
wget https://dl.sd-card-images.johang.se/debians/2025-02-10/ubuntu-oracular-arm64-shah4y.bin.gz
zcat boot-odroid_m1.bin.gz ubuntu-oracular-arm64-shah4y.bin.gz > sd-card.img
dd if=sd-card.img of=/dev/sdXXX # ...where /dev/sdXXX is your SD card

These are the results of the oracular build. Same results though, neither of these boot.

The-Hiker, 2025-02-16:
Both builds use the same "boot-odroid_m1.bin.gz" so may be why neither boots?

The-Hiker, 2025-02-16:
Here are the results for the noble build:

wget https://dl.sd-card-images.johang.se/boots/2025-02-01/boot-odroid_m1.bin.gz
wget https://dl.sd-card-images.johang.se/debians/2025-02-10/ubuntu-noble-arm64-heich4.bin.gz
zcat boot-odroid_m1.bin.gz ubuntu-noble-arm64-heich4.bin.gz > sd-card.img
dd if=sd-card.img of=/dev/sdXXX # ...where /dev/sdXXX is your SD card

DDR Version V1.09 20210630
In
ddrconfig:7
LPDDR4X, 324MHz
BW=32 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=8192MB
tdqss: cs0 dqs0: 48ps, dqs1: -120ps, dqs2: -96ps, dqs3: -144ps, 
tdqss: cs1 dqs0: -24ps, dqs1: -120ps, dqs2: -72ps, dqs3: -168ps,                                                           
                                                                                                                           
change to: 324MHz                                                                                                          
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:41%                                                                                                 
dram drv:40,odt:0                                                                                                          
clk skew:0x62                                                                                                              
                                                                                                                           
change to: 528MHz                                                                                                          
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:41%                                                                                                 
dram drv:40,odt:0                                                                                                          
clk skew:0x58                                                                                                              
                                                                                                                           
change to: 780MHz                                                                                                          
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:41%                                                                                                 
dram drv:40,odt:0                                                                                                          
clk skew:0x58                                                                                                              
                                                                                                                           
change to: 1560MHz(final freq)                                                                                             
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                          
vrefinner:16%, vrefout:22%                                                                                                 
dram drv:40,odt:80                                                                                                         
vref_ca:00000071                                                                                                           
clk skew:0x26                                                                                                              
cs 0:                                                                                                                      
the read training result:                                                                                                  
DQS0:0x2e, DQS1:0x32, DQS2:0x32, DQS3:0x29,                                                                                
min  : 0xe  0xe 0x13  0xf  0x2  0x5  0x9  0x3 , 0xb  0x8  0x3  0x3 0x10  0xc  0xe  0xa ,                                   
      0x11 0x12  0xb  0xb  0x5  0x2  0x4  0x5 , 0xa  0x8  0x8  0x2  0xf  0xd  0xb  0xd ,                                   
mid  :0x27 0x28 0x2b 0x29 0x1c 0x20 0x22 0x1e ,0x25 0x23 0x1d 0x1d 0x29 0x25 0x26 0x23 ,                                   
      0x2b 0x2c 0x25 0x24 0x1e 0x1b 0x1d 0x1f ,0x23 0x1f 0x1f 0x1b 0x27 0x27 0x24 0x26 ,                                   
max  :0x40 0x43 0x44 0x44 0x37 0x3b 0x3c 0x39 ,0x3f 0x3f 0x37 0x38 0x42 0x3e 0x3f 0x3d ,                                   
      0x45 0x47 0x40 0x3e 0x38 0x35 0x37 0x39 ,0x3d 0x37 0x37 0x34 0x40 0x41 0x3d 0x3f ,                                   
range:0x32 0x35 0x31 0x35 0x35 0x36 0x33 0x36 ,0x34 0x37 0x34 0x35 0x32 0x32 0x31 0x33 ,                                   
      0x34 0x35 0x35 0x33 0x33 0x33 0x33 0x34 ,0x33 0x2f 0x2f 0x32 0x31 0x34 0x32 0x32 ,                                   
the write training result:                                                                                                 
DQS0:0x2f, DQS1:0xf, DQS2:0x13, DQS3:0xa,                                                                                  
[... cut ...]
Retrieving file: pxelinux.cfg/default                                                                                      
No ethernet found.                                                                                                         
Config file not found                                                                                                      
No ethernet found.                                                                                                         
No ethernet found.                                                                                                         
=>

monnier, 2025-07-30:
In your log, we can see that the machine boots using the on-board U-Boot (the one in the SPI).  the SPL that's in the SPI is supposed to look for a U-Boot on the SDcard and boot that, but apparently it doesn't follow the same protocol as the rk3568 itself, so the U-Boot needs to be installed differently for this to work.

See also https://forum.odroid.com/viewtopic.php?f=211&t=50246&p=399119&sid=0051d63121a806afb11a794d66b3fd02#p399119 where I'm trying to figure out what's the magic form the SDcard needs to have for this hand-over to happen.

In the mean time, you should be able to boot @johang's image if you press the "boot select button" while booting up, so as to circumvent the SPI.  It works for me.