johang/sd-card-images #177: Can't boot Debian Trixie on Odroid HC4.

dwojewod, 2025-04-05:
Build date: 2025-03-31 

Odroid HC4
Petitboot 20220317

sd card was detected and I was able to list files on both partitions:

mmcblk1p1 mounted at /usr/var/petitboot/mnt/dev/mmcblk1p1 
mmcblk1p2 mounted at /usr/var/petitboot/mnt/dev/mmcblk1p2

[Disk: mmcblk1p2]  NO LABEL
Boot Menu:
Kernel: /boot/vmlinuz-6.12.20-arm64
Initrd: /boot/initrd.img-6.12.20-arm64
Device tree: <empty>
Boot arguments: root=PARTUUID= rw rootwait

After "Performing kexec load":
No signal detected on monitor.

I verified it on two different SD cards.

Debian Bookworm - no issue on the same sd cards.
DietPi also started.

darice11, 2025-06-27:
> johang left a comment [(johang/sd-card-images#177)](https://github.com/johang/sd-card-images/issues/177#issuecomment-3009422020)
>
> Connect to the UART console and show the complete log output during boot.

I believe you have used one of the Trixie kernels and not a Odroid compiled one. I did the same with same results. Had to go back to using the bookworm kernel with the Odroid addons. Need to wait until they get incorporated into Trixie or do a manual compile.

Trixie works fine with the 6.12.0 Odroid bookworm kernel.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

darice11, 2025-07-27:
ok, loaded the N2 trixie image on a N2 box I have and it worked first time. Went back to the HC4 Trixie images and still the same thing. Blue light stops flashing and it doesn't boot. which is unfortunate :(

monnier, 2025-07-30:
I suspect that some minor adjustment to the `/boot/boot.cmd/src` file should be sufficient to make it boot via Petitboot.  Or making the filesystem image use `u-boot-menu` might also help.

monnier, 2025-07-30:
I think it's because your `boot.cmd` script doesn't specify any DTB file.  IIUC Petitboot tries to parse the `boot.scr` (or `boot.cmd`?) and presumably looks for a few `load`s followed by `(z)boot`.
Looking at the `boot.cmd` you generate in `2nd-stage-files/pre-2nd-stage-files-*/etc/initramfs/post-update.d/zz-update-uimg`, I wonder: how does it work?  how does the vanilla kernel find the right DTB?

johang, 2025-07-30:
> I wonder: how does it work? how does the vanilla kernel find the right DTB?

Kernel doesn't find the dtb. It's U-Boot that loads the dtb. `zz-update-uimg` generates a boot script with variables that U-Boot fills in at boot time.

monnier, 2025-07-30:
That's indeed what I guessed, but I can't see where that happens in `zz-update-uimg`.  
Also, does that mean that it works only if U-Boot contains a "full" (and up-to-date) DTB?
[ BTW, the first comment in `zz-update-uimg` still says "Generate initrd.uimg ..." whereas it looks like you don't do that any more.  ]

johang, 2025-07-31:
In `2nd-stage-files/pre-2nd-stage-files-arm64/etc/initramfs/post-update.d/zz-update-uimg` you can see that it generates a script that searches for dtb in a few known directories. U-Boot doesn't need to contain a dtb because it searches for a dtb among the dtbs shipped by Debian.

monnier, 2025-07-31:
What I meant starts with the patch below:

```diff
diff --git a/2nd-stage-files/pre-2nd-stage-files-arm64/etc/initramfs/post-updat\
e.d/zz-update-uimg b/2nd-stage-files/pre-2nd-stage-files-arm64/etc/initramfs/po\
st-update.d/zz-update-uimg                                                      
index e72f56c..ca07d30 100755                                                   
--- a/2nd-stage-files/pre-2nd-stage-files-arm64/etc/initramfs/post-update.d/zz-\
update-uimg                                                                     
+++ b/2nd-stage-files/pre-2nd-stage-files-arm64/etc/initramfs/post-update.d/zz-\
update-uimg                                                                     
@@ -6,12 +6,8 @@ ABI="${1}"                                                     
                                                                                
 # List of paths to search for fdtfile in                                       
 FDT="\                                                                         
-/lib/firmware/${ABI}/device-tree/\${vendor} \                                  
-/lib/firmware/${ABI}/device-tree \                                             
 /usr/lib/linux-image-${ABI}/\${vendor} \                                       
-/usr/lib/linux-image-${ABI} \                                                  
-/usr/lib/linux/\${vendor} \                                                    
-/usr/lib/linux"                                                                
+/usr/lib/linux-image-${ABI}"                                                   
                                                                                
 # Create U-Boot boot script                                                    
 cat << EOF > /boot/boot.cmd                                                    
```

Because, AFAICT the image does not contain any `/lib/firmware/` directory nor any `/usr/lib/linux/` directory.
So it seems the loop serves only to test whether the `${fdtfile}` we receive from U-Boot already contains `${vendor}`.

darice11, 2025-08-02:
> johang left a comment [(johang/sd-card-images#177)](https://github.com/johang/sd-card-images/issues/177#issuecomment-3146366022)
>
> No, can't remove the paths. Ubuntu uses different paths.

I was reading up on how armbian linux works and noted that once the petiboot code is bypassed, the fan stops working and the cpu can over heat. Ran “stress” on it and the fan is definitely not kicking in.

armbian’s solution is to install a program “fancontrol” but that program says there are no fan-capable sensors modules installed. This is on your build johang where we bypass petiboot. Tried fancontrol on the linuxfactory build and it works fine.

I would suggest that until a fan control solution is installed the build may be a bit dangerous for the cpu. Unfortunate as I quite liked the build. Linuxfactory is not uptodate with their kernel builds.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

monnier, 2025-08-03:
@darice: I suggest you make it a separate issue.  My guess is that the problem is either in the vanilla U-Boot or in the vanilla Linux kernel (whereas Linuxfactory probably uses local patches by Hardkernel) and not in @johang's way to build images.

monnier, 2025-08-04:
BTW, I can confirm that replacing the `for+if` loop in the `boot.cmd` with just the inner `load+booti` for the one directory that contains the DTB allow Petitboot to boot properly.